betaflight/betaflight #2579: Preview in receiver is not updated

kingold5, 2017-03-06:
I use Spektrum DX6, CC3D FC and a receiver v710 from V-technology, not original from Spektrum but it is compatible with Spektrum TX when I select PWM receiver mode in configuration.
I trim the mid point value to 1496, which is the closest value to 1500 that I can trim. After I set mid_rc=1496 and deadband=3, the drone in receiver preview still rotates. I tried large deadband, different mid point, the drone rotates the same rate, even when mid_rc is 1300. Is the Preview regardless of mid point and deadband?

mikeller, 2017-03-06:
It is a known limitation of the configurator that the mid point setting is not taken into account for the 3D model display. For now, when adjusting the mid point, just disregard the 3D model and work with the receiver channel bars.

kingold5, 2017-03-06:
Okay, thanks for information!

closed 2017-03-06 by mikeller