betaflight/betaflight #2584: Reorder Diff & Diff All Profile Outputs

Eganwp, 2017-03-07:
I'll make this code change tomorrow if I have time or someone else can, but we need to re-order the DIFF & DIFF ALL dumps for each profile to be listed in the same order as Configurator as it's really annoying when comparing dumps to GUI values being out of order.

Reorder diff output to match BFC:
Roll pids
Pitch pids
Yaw pids

From what it is currently:
BFC:
Roll pids
Pitch pids
Yaw pids

Diff output currently:
Pitch pids
Roll pids
Yaw pids

- Travis

martinbudden, 2017-03-07:
Please wait until PR https://github.com/betaflight/betaflight/pull/2578 has been committed before you do this work. That PR makes substantial changes to the CLI code.

Eganwp, 2017-03-07:
Sounds good Martin. Thanks