betaflight/betaflight #2588: bf 3.1.7 / 3.2.0 bricked revolt fc

TheAngularity, 2017-03-07:
wanted to test bf 3.1.7/3.2.0 ... but after flashing and reboot my revolt is bricked and beeps 4x short and 2x long ... back to 3.1.6 with shorting boot pads all is good again ... any suggestion what's wrong :((

mikeller, 2017-03-07:
Tried flashing with 'Full chip erase' to make sure it's not a messed up config?

TheAngularity, 2017-03-07:
flashing the revolt with option 'Full chip erase' is not working

AndersHoglund, 2017-03-07:
SSG not detected properly??! , https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-does-my-flight-controller-blinkbeep-lots-of-times-when-powering-up-

My Revo with MPU6000 starts up just fine on 3.1.7

    # version
    # Betaflight / REVO 3.1.7 Mar  7 2017 / 11:06:19 (05f5a18)

TheAngularity, 2017-03-07:
tested now latest `3.2.0 (c864031)` ... but blinks and beeps

AndersHoglund, 2017-03-07:
ICM20602 aka "SSG".

borisbstyle, 2017-03-07:
@AndersHoglund There is a change made in gyro detection what may break it. 

It seems to work here locally on different board
```# status
System Uptime: 13 seconds
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM20608G, ACC=ICM20608G```

TheAngularity, 2017-03-07:
@AndersHoglund  @borisbstyle 
investigated now and found the prob in this booth commits 2993827 and 8c13087

borisbstyle, 2017-03-07:
There is issue in there after saving (warm reboots)

The boards I have here work on cold reboot

borisbstyle, 2017-03-07:
i reverted the commit that caused this till it is ironed out

closed 2017-03-07 by borisbstyle