betaflight/betaflight #2592: Multiple UART1 ports displayed after 3.1.6 upgrade

carloscampos775, 2017-03-07:
After upgrading my SP Racing F3 EVO FC to 3.1.6 the configurator is now displaying multiple UART1 ports 

[image]

mikeller, 2017-03-08:
@jflyper: There seems to be a problem in how the data structures for softserial port are initialised:

```
CLI

# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
```

(The `0` identifier is the same with the SOFTSERIAL feature enabled or disabled.)

TheCluster, 2017-03-08:
In 3.1.6 software serials for SP Racing F3 EVO was disabled, but counter of serial ports wasn't changed.

mikeller, 2017-03-08:
@TheCluster: You're right, good spot.

Also, this code is confusing to say the least:

```
#define USE_SOFTSERIAL1
#define USE_SOFTSERIAL2

#define SERIAL_PORT_COUNT       6

#define USE_ESCSERIAL
#define ESCSERIAL_TIMER_TX_HARDWARE 0 // PWM 1

#undef USE_SOFTSERIAL1
#undef USE_SOFTSERIAL2
```

jflyper, 2017-03-09:
I think SOFTSERIALx for EVO will be resurrected in 3.1.7.

https://github.com/betaflight/betaflight/pull/2575/commits/89e31d29e0cca6b34ca7d6fd2e03eeeedcf85fa2

mikeller, 2017-03-09:
@jflyper: Unfortunately I've had to disable it again, since the firmware wasn't fitting into the flash any more after merging some bugfixes.

jflyper, 2017-03-09:
😭

closed 2017-10-23 by jflyper