betaflight/betaflight #2594: KakuteF4 SBUS and Smartport not work with BF3.1.6

jamming, 2017-03-08:
There are no hardware conversions on kakuteF4. Sbus signal from frsky receiver is directly connected to UART3-RX. In betafilght configurator, I can not  receive any RC chanel. I have try both " set  subs_conversion = ON/OFF" in CLI. It still not work.
After adding an hardware invertor(SN74LVC1G04DBVR) between  frsky receiver  and UART3-RX, Sbus work well. 
So we must add a  hardware invertor for sbus port on F4  FC board?  As I know,there is no hardware invertor  on spracingF3.

mikeller, 2017-03-08:
Hi @jamming. Yes, for F4 MCUs, a discrete inverter is required to be able to use inverted protocols like SBUS or SmartPort. This is because, unlike the F3 MCU, the F4 does not have builtin inversion capabilities for UARTs. Betaflight does support switchable inverters (often an XOR gate is used for this), they can be switched in software `sbus_inversion`, and have to be connected to the pin indicated in `target.h` with `#define INVERTER_PIN_UART1 <pin>`.

jamming, 2017-03-08:
Hi Keller, 
I see ,Thanks !

borisbstyle, 2017-03-08:
Try softserial

jflyper, 2017-03-09:
I wouldn't recommend software serial for sbus... :eek:

closed 2017-03-20 by jamming