betaflight/betaflight #2602: bf 3.2.0 - connect to configurator broken

TheAngularity, 2017-03-09:
@martinbudden 
your latest #2578 breaks the connection from my revolt f4 to configurator ... with commit before all is good

TheCluster, 2017-03-09:
I have same problem with BluejJayF4

martinbudden, 2017-03-09:
Thanks for the report. We're looking into it.

DieHertz, 2017-03-10:
After ruling out that this issue is not related to Parameter Groups, I've performed the good-ol' `git-bisect` and found the culprit: https://github.com/betaflight/betaflight/commit/447325ec2b023e8f9c4e344166b5fd40315894d3
Some boilerplate code is missing, which broken it for my PIKOBLX (F3 VCP), and probably is the reason for other VCP boards.

DieHertz, 2017-03-10:
Structures required to execute `usbGenerateDisconnectPulse()` are not yet initialized, hence the call to `IOGetByTag(...)` return some rubbish most likely.
@mikeller suggests returning the old boilerplate code for now, without using `IOGetByTag()` if I'm correct.

DieHertz, 2017-03-10:
Actually it turned out to be much easier :-)

DieHertz, 2017-03-10:
Issue with F4 is a different one, still reproducible on my SOULF4 which doesn't connect to configurator altogether, even on UART3 via FTDI (MSP was enabled on this port by hard-coding).

martinbudden, 2017-03-10:
@TheAngularity , @TheCluster , can you confirm this fixes your problem?

MJ666, 2017-03-10:
AF F4 and NucleoF7 also does not connect anymore with latest master.

TheAngularity, 2017-03-10:
@martinbudden 
no :(( ... using latest master but not fixed yet

martinbudden, 2017-03-10:
So the F4 issue still needs to be tracked down.

TheCluster, 2017-03-10:
@martinbudden no. latest master and bug not fixed yet.

TheAngularity, 2017-03-13:
something new ... actually I test latest master `0f19256` but now the fc does not even initialize anymore

martinbudden, 2017-03-13:
It possible that PR https://github.com/betaflight/betaflight/pull/2496 cause the problem.

martinbudden, 2017-03-13:
@TheAngularity , @TheCluster , can you try PR https://github.com/betaflight/betaflight/pull/2643 to see if it fixes the problem.

martinbudden, 2017-03-13:
Sorry, I missed the comment about 0f19256, new PR reverts both suspect changes - https://github.com/betaflight/betaflight/pull/2644

DieHertz, 2017-03-14:
Please check it now after #2651 has been merged

TheCluster, 2017-03-14:
@DieHertz it works, thanks. Tested on BlueJayF4

DieHertz, 2017-03-14:
Awesome!

TheAngularity, 2017-03-14:
@martinbudden @DieHertz 
yeap ... i can also confirm it is running again ;-)) ... i think we can close because everything is working

martinbudden, 2017-03-14:
Closing, fixed by PR https://github.com/betaflight/betaflight/pull/2651

PR #2610 (2017-03-10): Fixed USB disconnect (pull LOW for 200 ms)
New code was pulling PA12 HIGH instead of LOW, tested on PIKOBLX after fixing. Tests on BlueJay and other VCP targets are welcome.

Fixes #2602.
closed 2017-03-14 by martinbudden