betaflight/betaflight #2607: COLIBRI_RACE fails dshot if resource map for motors is changed

mdrjr, 2017-03-09:
I have a colibri_race where the PWM4 is dead somehow.
Moved the motor to PWM6 and use resource motor to change the map.

When using DSHOT600 the remapped motor won't work.
Works with Multishot, OneshotXXX.

Tested mapping pwm3 to use pwm5 pin and the same issue happens.

Tested on BF 3.1.5 as 3.1.6 is broken.

jflyper, 2017-03-11:
DMA channels other than motor 1~4 are not defined for this target.

The definitions in the old format, in which DMA channels must be defined explicitly. Converting them to new format which derives DMA channels automagically may lead to usable definitions. Why don't you ask the target maintainer (== TBS) for this mod?

And please not that resource remapping capability does not ensure you can arbitrary move around functions. It works under certain contraints, and for Dshot, the constraint is very strong.

mdrjr, 2017-03-13:
Well, since it depends on TBS to fix it, lets close this issue as I have no expectations that they will ever fix it. See #256 A 1 year old issue waiting on TBS.

closed 2017-03-13 by mdrjr