betaflight/betaflight #2609: Feature Requests - OSD

SlipstreamFPV, 2017-03-10:
Would love to see the following updates on the OSD:

-Option to turn off leading and trailing characters per item. (e.g. turn off mah symbol before mah drawn readout, turn off 'fly' before timer etc.)

-Option to have Amps readout limited to one decimal place (e.g. 10.2 rather than 10.23)

-If OSD items have a trailing symbol then the characters should be right aligned

DanNixon, 2017-03-10:
An easy solution for the first issue is uploading a font that has blank characters for the symbols you don't want.

SlipstreamFPV, 2017-03-12:
@DanNixon I already did that before I posted but thanks for your reply! One example of why that fix isn't great is that the "V" that comes after voltage is the character letter "V" not the character for volts which is included in the MCM file. So if you blank that out then you get no "V" anywhere else in the OSD either.

riquezjp, 2017-07-24:
I was going to suggest a similar thing. I'd rather see Amps with no decimals on the OSD.
When we are flying mini-quads at least, Amps will go from 20 to 80 in under a second the .34 is pointless. 
I just want to concern myself with noticing if I will reach the ESC limit 100A or 120A etc.