betaflight/betaflight #2612: Throttle 1900 and above leads to a reset while stuck in a full deflection roll.

gpwelter, 2017-03-10:
Throttle 1900 and above leads to a reset while stuck in a full deflection roll. If the throttle is left at 1900 or above and a full deflection roll it will continue this behavior with regular intervalled resets. Currently using uart 3, revo f4, behavior does not happen with an f1 board. Board does not have to be powered to produce this result. Currently using ibus but have seen it using sbus. Can be seen using the receiver previewer in betaflight. This is an sbus version of this issue https://www.youtube.com/watch?v=k3RYl3KBAvk. I currently circumvented the issue by setting my upper endpoint below 1900.

bonhardt, 2017-03-11:
I can confirm the same issue.
When Throttle + Roll together is over 1900, serial link resets, buzzer beeps once from FC

Omnibus F4 V2
IBUS
BetaFlight 3.1.5

ZeeZeeEff, 2017-03-11:
Hey, I am having problems with "desyncs" but maybe it is this problem right here!! This is causing my ESC's to reboot

watch here: https://youtu.be/bifr2XZwecE

My setup:
FC: AIRBOTF4
- mapped solderpads near MCU for softserial, using it for smartaudio
ESC: SPEDIX 30A BLHeli_s 16.6
RX: Frsky XSR on SBUS

how can we help to debug this?

gpwelter, 2017-03-11:
This problem was addressed in an earlier reported issue and listed as a stick command #2576 5 day ago. Raise max_check = 2000 and set endpoint to below 2000. This is a work around but does address the issue. It will do this whether or not you have anything hooked to the board. I test boards before attaching everything to them and saw the behavior then but thought it to be a power issue and assembled it with the same results. Not a desync issue but stuck command issue.

mikeller, 2017-03-12:
@gpwelter: Is there still a problem or can this issue be closed?

gpwelter, 2017-03-12:
Not sure if it is an issue. I don't believe the behavior is in line with other versions of software or hardware that I have come across. If a work around is the solution should it be closed?

mikeller, 2017-03-12:
So just to recap:

The behaviour you are reporting is that the flight controller resets depending on stick position when disarmed? If so, this is not a bug, but a feature, it is called 'stick commands': 

https://github.com/cleanflight/cleanflight/blob/master/docs/Controls.md

(Note that some of these commands, like profile changes, will cause the flight controller to reboot.)

If this is the behaviour that you are reporting, then this issue should be closed.

gpwelter, 2017-03-12:
This would not be so. This occurs during flight as well but can be worked around by raising max_check to 2000 and lowering tx throttle endpoints below 2000. A work around is not a solution. I never had any issues with in flight stick commands until I jumped to the f4 board. The other stick commands do not interfere with flight to the best of my knowledge.

mikeller, 2017-03-12:
Yes. If this happens when armed it definitely is a bug. Stick commands should only be active when disarmed.

hydra, 2017-03-12:
hehe, except the disarming one 😄

gpwelter, 2017-03-14:
'

gpwelter, 2017-03-14:
Not satisfied with the explanation other 2 F1 boards(naze32 and CC3D) do not behave in this manner yet is attributed to stick command. Receiver does not have telemetry as my others do.

mikeller, 2017-03-14:
Check your settings for `max_check`, and the maximum channel values that your RX can reach. If the RX cannot go above `max_check`, stick commands will not be recognised.

gpwelter, 2017-03-14:
Correct. This was the work around discussed 2 days ago. I will have to see how the rx works with the F1 boards.

closed 2017-03-15 by gpwelter