betaflight/betaflight #2613: Change camera settings with betaflight osd

michael-blue2, 2017-03-10:
Is it possible for a next feature to connect our cameras with the betaflight osd so we will be able to change the camera settings from the betaflight osd or taranis, without having to open up our quads and connect our camera with the osd setup cable

jflyper, 2017-03-11:
You want access your cam's osd control signal from FC?
Similar function has been brought up couples of times in the past.
What is your cam? And how did you know it's uart?

Arakon, 2017-03-11:
As far as I know, most cameras use I2C or SPI to control their setup.

jflyper, 2017-03-11:
@Arakon I've never seen such a beast! Pointers?

TheCluster, 2017-03-11:
Most popular cameras (such as foxeer hs1117,aomway, runcam) are use analog control signal for camera menu. OSD board for my foxeer XAT600M/XAT1200M cam hasn't any chip, but have 5 buttons and array of resistors.

michael-blue2, 2017-03-11:
i know its analog signal, and all the cameras that i know are using that 5 button or the nob that moves up/down/left/right. As you have already mentioned most cameras are using I2C which gathers the analog signal from the osd board. So by that you mean that there is no any way of this to work?

jflyper, 2017-03-11:
You may be able to do with PWM coupled with RC filter 😁

michael-blue2, 2017-03-11:
with pwm? maybe its the only way doing it as our cameras are analog yet and getting a pupse signal from the osd control cable

jflyper, 2017-03-12:
@michael-blue2 
Use 1 PWM output with a RC filter. The RC filter rectifies the PWM wave form into voltage level that reflects duty ratio of the wave form. Anyone willing to do some experiment? :)

mikeller, 2017-03-12:
@jflyper, @michael-blue2: Or use one of the DACs present on the STM32 chips.

jflyper, 2017-03-12:
Wasn't DAC hard to use? I thought it's kind of fixed pin-wise...

mikeller, 2017-03-12:
Yes, they seem to be shared with SPI1 SCK / NSS for both F3 and F4. :-( Why, STM, why?

Cheredanine, 2017-03-14:
Just an observation guys, believe this functionality is already implemented in tbs core pro, whilst that doesn't help particularly, it does at least indicate is isn't rocket science

marcroe, 2017-03-16:
What about a digital potentiometer in rheostat mode that mimics the resistances from the osd cable? The potentiometer could be connected to the FC via I2C, SPI, or similar.

jflyper, 2017-03-16:
@marcroe We can do dozens of things if we can add a circuit.

DieHertz, 2017-03-17:
I'm currently working on a way of controlling HS1177-style OSD via a spare GPIO, it'll require user to build a simple RC filter like the ones used for PWM RSSI concersion to analog voltage.
I hope to create a PR soon.

jflyper, 2017-03-17:
UI !?

DieHertz, 2017-03-18:
A sneak peek, currently using Arduino, going to port to STM32 tonight or tomorrow :-)
https://youtu.be/mtNUbYJ27TU

DieHertz, 2017-03-22:
Check out the related PRs

michael-blue2, 2017-03-22:
Will this be able only from an extra motor output?
Most fcs have 4 motor outputs as the bf fc aswell

DieHertz, 2017-03-22:
Far from most, many FCs have more than 4 outputs for motors, however this feature requires not a motor pin but a pin with free timer channel.

DieHertz, 2017-07-29:
Feature is in master for a while and works for many people. There's more to be done, e.g. rework software PWM version to hone it for F4/F7 and also allow hardware PWM timer sharing, but it'll take some time.

closed 2017-07-29 by DieHertz

GrishaLutsenko, 2017-09-03:
I am not sure if asking in a right place, but...
For foxeer it looks to be working fine, but runcam cameras require bit higher voltage on control pin.
For entering menu it requires 1.5 volts on OSD pin, I tried changing camera_control_ref_voltage, but that gives 0.6-1 volt range, can it be rised ? Or maybe adding runcam as option with different PWM values ? If required I can measure required voltages for all buttons...

DieHertz, 2017-09-03:
I have implemented and tested this functionality using RunCam cameras.
They use 3.2-3.4V reference with resistor values of 45k/27k/15k/6.8k/0 vs 47k inside the camera, just like _every_ other camera on the market. That gives `3.3 * 45/(45+47) ~= 1.6V` for entering the menu. Something is wrong with your setup.

GrishaLutsenko, 2017-09-03:
I use 470 Ohm between FC and camera. I see PWM output on oscilloscope and there is about 30% duty cycle, that matches to voltage I get after resistor.

DieHertz, 2017-09-03:
At which point are you scoping into the signal, before the resistor (FC side) or after the resistor (Camera side)?
If you have an analog scope, watch the DC value on the camera side as well.

Which FC you're using, how did you set up the `camera_control` feature and which camera is it?

DieHertz, 2017-09-03:
@GrishaLutsenko firmware outputs PWM with duty cycle of `cameraRefVoltage / adcRefVoltage * 45K / (45K + 47K)` for `enter`, which is roughtly `49%`. If it outputs something else then there is an issue elsewhere, there should be 49% duty cycle at `camera_control` designated pin.
Which target your FC uses, which resource did you select for `camera_control`, is it `software_pwm` or `hardware_pwm`?

GrishaLutsenko, 2017-09-03:
Hm... About feature I am trying now but it don't shows me it at all. 
But definitely I have output on assigned pin and it is different according to camera.

So what I see is when throttle in MID and I press roll left -  I get 1.6 v (and 50% duty cycle) but duration is not enough to enter OSD. If I enter OSD with joystick I can navigate down in list, nothing else works.

When throttle is LOW and I push pitch UP, then I get about 30% duty and see 1 volt after resistor.

Camera is Runcam Swift 1

DieHertz, 2017-09-03:
You still didn't tell me how you set it up in software and at which point you're scoping it.
This matters greatly. RunCam Swift was one of the cameras I used to develop this feature.

GrishaLutsenko, 2017-09-03:
I set resource CAMERA_CONTROL 1 A08 (of course before that I cleaned resource, there was motor 6)

After that I started to get PWM output on that pin depending on stick positions.

DieHertz, 2017-09-03:
Please show `resource list` command output as well.

GrishaLutsenko, 2017-09-03:
Entering CLI Mode, type 'exit' to return, or 'help'

# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource LED_STRIP 1 A01
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource CAMERA_CONTROL 1 A08
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: LED_STRIP
A02: MOTOR 4
A03: MOTOR 3
A04: MPU_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FLASH_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: INVERTER 1
C01: FREE
C02: ADC_BATT
C03: FREE
C04: MPU_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: LED_STRIP
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Use: 'resource' to see how to change resources.

GrishaLutsenko, 2017-09-03:
I will send you oscilloscope screen now too (for Vref=330, pressing enter)

DieHertz, 2017-09-03:
RunCam cameras register clicks after 125ms, so including settling time you shouldn't need more than 150-180ms.
`Vref` should be set to the value measured between camera OSD and GND pins when nothing is connected to them.

GrishaLutsenko, 2017-09-03:
There is 3.15 volts when nothing connected, will fix now. Ok, now I can navigate UP too.
But entering is still problem.
Please check picture, that duty cycle can't do 1.6 volts definitely http://radikal.ru][img]http://s018.radikal.ru/i518/1709/43/9e9aa388590a.jpg

Also, I checked how enter works in menu, it does same as UP button.

DieHertz, 2017-09-03:
Is this picture taken by scoping FC side of the resistor or Camera side?

GrishaLutsenko, 2017-09-03:
FC side of course. On camera side is stable voltage

GrishaLutsenko, 2017-09-03:
What FC you tested with rancam ? Maybe there is issue around F4 board ?

DieHertz, 2017-09-03:
Different FCs, BlueJayF4, CLRacing F4, Matek F405, some F7s.

GrishaLutsenko, 2017-09-03:
Hm. So you see graph. Actually now UP and DOWN work correct. Right and left also make change in menu. But for enter I get 30% duty cycle and that is wrong. 
Btw, have you checked RC4 or some other build ? Maybe in RC4 there is bug around that ?

DieHertz, 2017-09-03:
[image]
Yes I have. Please check this infographic and make sure you're using correct stick command for `enter`.
It's highly unlikely that everything but ENTER works unless you're on `SOFTWARE_PWM` which has issues with duty cycle close to 50%.

GrishaLutsenko, 2017-09-03:
Oh, you made me feel idiot, looks I used older infographic where throttle was down and then pitch up for enter.... this way works amazing, thanks you very much.

Btw, why it sends UP when throttle is at 0 and pitch UP ? Not fully removed older variant ? It definitely does it with enter duration, but with UP value

DieHertz, 2017-09-03:
Some weirdness in Stick Commands implementation I presume. I plan on reworking them to minimize delay with modern protocols like SBUS, I'll look into this issue as well.
Glad it works for you now!

GrishaLutsenko, 2017-09-03:
Thanks again. This is amazing feature as otherwise that was a pain to setup camera correctly. I think I also made my BF fly similar to RF, hard play around settings, pids and TPA.... A bit more and think I will achieve same level of control. And if it will fly same + all benefits in BF that will be amazing...
Sorry for offtop :)

DieHertz, 2017-09-03:
I have started working on a wiki page, hopefully soon everyone will be able to enjoy it :-)
https://github.com/betaflight/betaflight/wiki/FPV-Camera-Control-(Joystick-Emulation)

AliB10, 2017-09-05:
got this working on foxeer arrow v3. had to drop camera_control_key_delay =  to 125 to get enter to work reliably. working perfect now though. thanks. awesom feature to have in OSD.

GrishaLutsenko, 2017-09-13:
Sure you released B07 first from motor output ?

mikeller, 2017-09-13:
@GrishaLutsenko: Releasing the pin is not necessary, motor 5 will not be used if the mixer for a quad is selected. (But releasing the pin won't hurt either.)

But have you done `resource camera_control b7`, @iceberg101?

GrishaLutsenko, 2017-09-13:
@mikeller I checked on mine F4 board and until I released pin from motor 5 output I had no output for camera there. But for LED strip that works without releasing. Not sure why, just observed that.

iceberg101, 2017-09-14:
Yes I have re mapped motor own pin to camera control.

iceberg101, 2017-09-14:
Yes I have re mapped motor 5 pin to camera control. Previously I tried on led pin (A08) where i also alocated the resource for camera control. I've also tried changing the key delay from 100 to 500 in increments of 10 with no succes.

Drewlrox, 2017-09-14:
Hello, so I see in the wiki it reccomends to run an extra ground. Is this talking about camera gnd or signal gnd wire? I already have my camera grounded to my fc for my bf osd, should I run another from the signal gnd wire?

iceberg101, 2017-09-14:
The ground referred to here is from the osd wire

Drewlrox, 2017-09-14:
So I just wire that to any spare gnd pad on my fc?

iceberg101, 2017-09-15:
That's correct

noypi2000, 2017-09-16:
I have an issue with my Monster V2 and osd camera control. I have Omnibus F4 V2. Used 220 resistor, 125 delay and 99 internal resistance. I tried motor pin 5 and 6 with same result. I can only go to the 1st menu using joystick combo yaw low and pitch up. If I change the internal resistance 50 using the same stick combo I can get to the 2nd menu but I cannot move up, down, left or right.  I also tried the LUA script with the same problem.

DieHertz, 2017-09-16:
I can't help you as I don't have this camera

supiiik, 2017-09-19:
Same problem here with monster V2, omnibus F4 V1 (BF 3.2.0 RC5). joystick has standard resistors 0, 6.8, 15, 27, 45ohm, Voltages for keys are here: 
up - 1,95V
down - 0V
left - 2,37V
right - 1,31V
Enter - 2,66V
From these values it seems, that internal resistance is 10,7kohm

I can't get 1st (need long press up) or 2nd menu (short press enter), but when I get it by the joystick, then I can move in 1st menu (name, voltage,...) all the ways and also confirm by the transmitter. In the second menu I can move only down (0V)
I was trying to change almost everything. The resistance seems to be correct at 107 (tested from 10 to 1000), 1st menu working properly from 75 to 140, so the middle value is also around 107. The delay is working from 100 to 500, but the higher number results in repeated menu movements. Value 100 seems to be fine. 
I'm using 300ohm resistor without capacitor at A08 hardware PWM.
Is there something else, what I can try?

GhostMaster9, 2017-09-20:
I cant make it work with the Foxeer Micro Arrow, very odd and strange... 

Tryed several resistor values and settings. With Runcam Micro swift is straight without any resistor...

michael-blue2, 2017-10-08:
it works with 2 out of 3 of my board. kakutef4 aio with hardware pwm and 300ohm resistor, lux fc with hardware pwm aswell and the same resistor. but i cant get it to work with kiwiv2 fc. i tried led strip output and motor output as well. i will add a cap as soon as possible and give it a try with software pwm

michael-blue2, 2017-10-14:
NEWS:
It does work with the kiwi v2 with software_pwm but i had to increase a bit the voltage. with my measurement my camera_control_ref_voltag had to be 328 but with that value it wouldnt register some of my kews which i first decreased it(loosed every command) and after i increased it until 343 which is working fine 👍

[3 more replies]