betaflight/betaflight #2615: Config diff does not work on master

unitware, 2017-03-10:
Checked out master and built and flashed with full chip erase on a spracingf3neo

- When clicking restore defaults in configurator the buzzer goes failsafe "SOS" and the configurator lost contact with the quad.
- Powercycling a few times i have contact again.
- Trying out the cli diff command, same result (SOS, lost contact)
```
CLI

# diff
# Betaflight / SPRACINGF3NEO 3.2.0 Mar 10 2017 / 08:45:15 (4e39aa2)

rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h
VALUE sdcard_dma ERROR
VALUE vtx_band ERROR
VALUE vtx_channel ERROR
VALUE vtx_mode ERROR
VALUE vtx_mhz ERROR
```
- applying the cli diff from before fails after/during "set motor_pwm_protocol = MULTISHOT
" 
```
# diff
# Betaflight / SPRACINGF3NEO 3.1.6 Mar  9 2017 / 20:13:32 (eec32ac)

feature -RX_PPM
feature RX_SERIAL
serial 1 64 115200 57600 0 115200
serial 4 4096 115200 57600 115200 115200
set motor_pwm_protocol = MULTISHOT
set serialrx_provider = IBUS
set align_board_yaw = -45
set pid_process_denom = 2
profile 0

rateprofile 0
rateprofile 0
```

Currently the master is at
```
git log -1
commit 4e39aa2a966242f1a52494f4913f3b631b8c7555
Merge: 9b6bb82 d9cb02e
    Merge pull request #2610 from DieHertz/fix-usb-disconnect
    Fixed USB disconnect (pull LOW for 200 ms)
```

I'm using cleanflight configurator 3.1.1

martinbudden, 2017-03-10:
Partially addressed by PR https://github.com/betaflight/betaflight/pull/2616

unitware, 2017-03-14:
Tested today and this error has been fixed, thanks guys!
(master @ 6303de76500f23174412525756ded88e6a05a696)

closed 2017-03-14 by unitware