betaflight/betaflight #2618: No receiver Input on F3

flare142, 2017-03-10:
I have a SP Racing F3 running 3.1.6, and on my receiver tab it doenst show any input signal from my transmitter in PPM or SBUS mode(all channels stay at 1500). I have all the wiring done correctly and settings. Is there a solution to this? The radio and receiver im using are  Futaba T10CAG and FM800 receiver .
Here are my settings maybe Im doing something wrong..

[image]
[image]
[image]
[image]

mikeller, 2017-03-11:
Just tested, works fine for me. But from looking at your picture, I suspect your wiring is wrong. The white wire from the RX should go into pin 3 not pin 4 of IO_2 for SBUS.

flare142, 2017-03-11:
I see thanks alot for the help, It was the included cable with the drone.

closed 2017-03-12 by mikeller