betaflight/betaflight #2619: OSD Font layout question/issue/feature request

elf128, 2017-03-11:
Hey,
First of all, sorry, I don't know where I suppose to post this and I'm not sure I have to post it as issue, cause it's more of a question. And I don't know, who I have to ask about that.

I'm implementing support of OSD for one of the boards. And, errr.. let me put it this way, it's not max7456 based.
So, to be as much closer to current OSD implementation, I'm trying to mimic functionality to make sure it will behave the same way as any other max7456 based board for a customer. And it all goes well, but I have a problem with font update.
Right now, as I can see from sources, font is not stored in BT firmware but it's pushed from configurator through MSP into max7456 directly.
So, here comes a problem. max7456 stores font on per character basis, 64 bytes per symbol. Each character is separate flash page, so you're not touching other characters while updating them one by one. In my case it's all line based and page is 1024 bytes long. That mean, information about character is scattered  around 18 different flash pages. Which mean if you want to update a single character, you have to obliterate entire ROM and write it over with old info, with one character changed.
Now, this is: a) SUPER inefficient, cause you'll rewrite flash 256 times if you want to update all symbols in the font. b) You have to waste at least 1K of RAM for page content while you reset the page.
So, here is a list of my questions:
 - How often you actually have to update font? Is it just for max7456 initialisation or it's really necessary?
 - How often BT configurator update just 1 symbol?
 - Can I be sure that, configurator will update symbol linearly, I mean, 0, 1, 2. Not in any other order.
 - Can I ask configurator to send me specific symbol, so I can ask for every character 18 times and reorganize data on BT side?
 - My problem could be solved easily, If I can ask configurator to organize font data differently and then send at once or split it into bunch of commands. Is it possible to reserve another MSP for that?

Again, sorry for making an issue out of it.

jflyper, 2017-03-11:
> How often you actually have to update font? Is it just for max7456 initialisation or it's really necessary?

At least once, probably a dozen times when the board is first arrived, then once in a while when a user feels being unlucky 😉 

> How often BT configurator update just 1 symbol?

Never (atm).

> Can I be sure that, configurator will update symbol linearly, I mean, 0, 1, 2. Not in any other order.

Currently, it does it linearly. Don't no about the future.

> Can I ask configurator to send me specific symbol, so I can ask for every character 18 times and reorganize data on BT side?

Do you mean requesting specific glyph to be sent from FC side? That doesn't happen in the current MSP client-server model.

 > My problem could be solved easily, If I can ask configurator to organize font data differently and then send at once or split it into bunch of commands. Is it possible to reserve another MSP for that?

Not likely. Extending the current MSP some how would be my choice.

elf128, 2017-03-13:
> Why don't you just begin with a big buffer in ram, receive entire glyph sets and send lines into your hardware? I think there is a room for that for non-F1 targets.
Well, locking down 18KB of RAM would be an issue even on F3. It's almost half of entire RAM :)

jflyper, 2017-03-13:
@elf128 Oh... 18KB is too huge...

I just looked into the FC side code for downloading (uploading?) font glyphs. The protocol PDU is
> <character address (1 * uint8_t)><glyph data (54 * uint8_t)>

We will find a way to handle another case later; either by adding a new MSP code or extending the existing one. You can go ahead and implement your line based hw & sw, both FC and BFC.

elf128, 2017-03-15:
Ok. I'll do it that way.
Thanks.