betaflight/betaflight #2620: OmnibusF4Pro barometer not working BF 3.1.6

mamaretti, 2017-03-11:
Hi, 
i have an OmnibusF4Pro and the Barometer does not work.
i tested with baro_hardware = AUTO  and baro_hardware = BMP280.
reflashed the Board doesn't help.
Are there any other options i have to change.
Please help, THX

jflyper, 2017-03-11:
@mamaretti Your board is broken.

`set baro_hardware = AUTO` should work.

[image]

mamaretti, 2017-03-11:
hi jfyper
i have  Board: OBF4
do you use  different firmware?
can you send a dump?
thx

mamaretti, 2017-03-11:
its working now.
i selected OMNIBUSF4SD instead of OMNIBUSF4

jflyper, 2017-03-11:
@mamaretti Have fun 😄

jflyper, 2017-03-12:
@mamaretti Can you close this issue before going flyin'? :)

closed 2017-03-12 by mikeller