betaflight/betaflight #2621: CC3D BF 3.1.6 - cant connect anymore

Noctaro, 2017-03-11:
Hi, 
i flashed my CC3D via betaflight configurator, connected via uart and arduino nano as ftdi. 
With 3.01 everything connected normal to gui. After flashing 3.1.6 i cant connect anymore. There seemed not to be a problem while flashing. If i try to reflash an older version i now get "no answer from bootloader". 
Did i do something wrong?

mikeller, 2017-03-12:
Can you try connecting to USB instead of serial? Everything should work fine that way. Support for configuration over serial port was disabled by default between 3.0.1 and 3.1.6.

Noctaro, 2017-03-12:
Thank you for the info, i really was wondering.
Thats a problem for my board, the usb was teared off. Can only connect via UART :/
Is there a way to revive it? Should i try STM Flash using an older firmware? Is there a way to enable UARTS in the newer revisions before flashing?

Noctaro, 2017-03-12:
Ah i saw this problem was already discussed here -> https://github.com/betaflight/betaflight/issues/2566

closed 2017-03-12 by Noctaro