betaflight/betaflight #2622: Sparky 2.0 Input

Tomcatt, 2017-03-11:
On the transmitter page in the configurator it shows all commands being received, but when armed, all I have is throttle control.  It's like the other inputs are being read.  And arm and disarm just fine.

mikeller, 2017-03-12:
Check your settings for `min_check` and `pid_at_min_throttle`. If `pid_at_min_throttle` is off, then RC inputs will only become active once your throttle is above `min_check`.

Tomcatt, 2017-03-17:
An older compile  of 3.2 I used gave me the same results.  HOWEVER, your latest compile of 3.2 I just compiled works!!  It fixed a throttle problem I was experiencing too.  Thank you.

I'm taking the quad to work this weekend and fly around there.  I'll leave feedback later

closed 2017-03-17 by martinbudden