betaflight/betaflight #2624: Beta 3.1.6 motolab tempest v1.3

emma0223, 2017-03-11:
There's seems to be a conflict when you turn the led strip on with dshot. motor 4 won't spin

mikeller, 2017-03-12:
That's well possible. Check the output of `resource list` in CLI. There should be an entry under 'DMA' for 'MOTOR 4' and one for 'LED_STRIP'. If not then they are probably trying to use the same DMA channel. You might be able to work around this by remapping motor 4 output to one of the motor 5 - 8 outputs with the `resource motor 4 <pin>` command. Easiest way to find out is to try.

closed 2017-03-12 by emma0223

emma0223, 2017-04-12:
@mikeller thanks it's was just as you explained  the dma where on the same channel for led and motor 4 ... forward motor 4 signal to motor 6 and everything is working now