betaflight/betaflight #2631: Quad randomly starts shaking when logging to sd, logging ends when shaking starts

bexamous, 2017-03-12:
So issue is when logging to sd card seemingly randomly quad beings to shake, seems like just on yaw axis to me but not positive.  It seems to continues to shake until you cut and re-apply throttle and when it stops quad flies just fine.

Blackbox log seems to end when shaking begins, but seeing file sizes it might just be getting corrupted and logging really is continuing.  Blackbox Explorer just shows log up til shaking, cmd line tool segv trying to decode.

The shaking is not desyncs, you still have control of the quad, its just shaking.

I've had issue happen a bunch, I've never had it happen twice on same pack though.  Not sure if that is luck, because blackbox is ending and that is what causes issue, or some other reason.

At least 5 people have this issue, see below.  Everyone is using Omnibus F4 Pro v2, OMNIBUSF4SD target.  I was using 3.1.0 but others have had issue with 3.1.5 and 3.1.6.

Here is a LOS video of issue and two blackbox logs:
https://www.rcgroups.com/forums/showpost.php?p=36929276&postcount=45406

Here is a FPV video and blackbox log:
https://www.rcgroups.com/forums/showpost.php?p=37030008&postcount=46108

Here is another FPV video:
https://www.rcgroups.com/forums/showpost.php?p=37032891&postcount=46137

Here is just two other people saying they have same issue:
https://www.rcgroups.com/forums/showpost.php?p=37039548&postcount=46176

https://www.rcgroups.com/forums/showpost.php?p=37050613&postcount=800

I've had issue reproduce using 8k as well as 2k.. I've not tried lower.  I can log on Omnibus F3 Pro at 8k without issue.  One person reported lowering logging rate and reformatting sd card fixed issue, but they didn't say how low that was.  If F3 board can log at 8k doesn't seem like F4 should be limiting to something under 2k.

mikeller, 2017-03-12:
Have you tried disabling SDCard DMA with `set sdcard_dma = off`?

jumatic, 2017-03-12:
Had same issue on my new build with Dshot 600, Omnibus F4 Pro V2, DYS XDS30A, 8K Logging. Shaking just appeared when i was switching to airmode for some seconds. Last flights with 4K Logging was ok.

Eganwp, 2017-03-12:
Is this not due to too high CPU utilization?

jflyper, 2017-03-12:
Is it meaningful to record at 8K?

sa-mendez, 2017-03-12:
I too have been running into this issue.  There is nothing scarier than to witness your quadcopter just bugger out randomly the way this causes it to.

Some observations/responses to the comments:

- **set sdcard_dma = off** works to eliminate the problem.  That is a suitable option.
- Another option is to drop the pid/gyro loop down to 4k (which also automatically drops the max log rate to 4k).  This has also worked for me, as I only experience this issue when running 8k pid/gyro loop time.
- Of course a final option is to just turn off logging completely.
- I have not witnessed any high CPU utilization in the logs when the out of control oscillations occur (at most 10-11% CPU load as reported by the debug log).  However,what I have witnessed is a massive spike in the Cycle Time.  It goes as high as 2300us on a normally 125us loop time.   Often times(but not always) this results in the logs getting cut off.

Is it possible that the PID loop is somehow getting stuck due to not getting a response from the SD card during logging?  This would explain why it suddenly rapidly oscillates as it tries to play catch up.  This is just a theory of mine and I dont even know if I am explaining it right ;-)

I can provide my logs upon request if anybody is interested.

Thanks,

Sergio

sa-mendez, 2017-03-12:
To follow up on my last post.  All the observations that I witnessed in the logs occur before the out of control oscillations occur.  From the moment the quadcopter starts behaving erratically, the logs completely cut off, and no more logging occurs until I disarm/rearm.

bexamous, 2017-03-12:
> Is it meaningful to record at 8K?

FWIW I've had this happen at 2k, 8k/8k with 1/4 logging.  The first link in OP the video and log is issue happening with 2k logging.    Maybe 8k logging isn't needed, but FWIW I can log at 8k with an Omnibus F3 Pro v2 without issue... seems odd the F3 board does it no big deal with 30% CPU or something.  The F4 board at 10% fails with 2k logging.

borisbstyle, 2017-03-12:
Maybe board has weak contact with the SD card?

sa-mendez, 2017-03-12:
At least for my board/setup, I do not see how this could be the case.  When I drop the pid/gyro/log loop to 4k, I have gotten complete logs every single time (from arm to disarm).  And I also verified that turning off sd_dma allows me to run using pid/gyro/log loop of 8k.  I only get the shaking behavior and log cut off when running 8k pid/gyro loop and sd_dma is ON.  If there was weak contact then I would have some 4k loop logs that would also exhibit shaking and log cutoff behavior.  I have yet to witness this.

compudaze, 2017-03-13:
Same issue for me.

https://youtu.be/CDCDbARuNaQ

Blackbox freezes and quad shakes like crazy.

Omnibus F4 v2 - BF 3.1.6
Kiss 24A - 1.08 DSHOT600

BB logging was at 2K. CPU was at ~5%.

titanrw, 2017-03-17:
I had the same thing on my Omnibus F4 Pro.  8k/8k with 8k logging, and on two different flights I got 2-3 seconds of hard shaking, but then it stopped.  Rest of the flight was fine.  Checked both blackbox logs later, and there were only partial logs.  I assume it stopped logging just before the shaking, but I honestly don't recall for sure.  Set logging down to 2k and never had it happen again, although I haven't flown it extensively since.  Still have yaw twitches at certain throttle settings, but that's a separate issue.  Have to try softmounting to fix that.

titanrw, 2017-03-18:
Here's a video of my two quads from today.  2k logging hasn't helped.  Each flight with an sdcard inserted for logging and it did it.  Every flight without sdcard was fine.   Haven't tried with sdcard dma turned off.

https://www.youtube.com/watch?v=ZPzaEDYmZh

googlebutt, 2017-03-21:
I believe I'm also having this issue, see my thread here for logs, video and setup info: https://www.rcgroups.com/forums/showthread.php?2855651-Quad-freaks-out-in-flight-at-mid-throttle%21

kamishams, 2017-03-21:
I'm having the exact same issue. Just switched to Dshot 600 - 8khz/8khz, on Omnibus F4 Pro v2 - Littlebee30A.
Blackbox log stop during the shakes. Gonna try the fix tomorrow.

HSPalm, 2017-03-31:
Maybe the developers like to review this "bug post".
https://blog.frankvh.com/2012/01/13/stm32f2xx-stm32f4xx-dma-maximum-transactions/

kev018, 2017-05-01:
Here the same issue.... only when i log the Blackbox Data to the SD File... at the first flight of the Day the quad shaking on the way to the Start Place, just a little throttle raise, than it starts shaking... at the same Time the Blackbox log stoped and the OSD refresh no Data. After that i removed the SD Card... 6 Lipo no problem. At the last Lipo with SD Card inside the same issue... look at the Video.  8k/8k, 1k Blackbox log DSHOT600. https://www.youtube.com/watch?v=ki5rBpCPyJA&feature=youtu.be

mikeller, 2017-05-01:
@kev018: Read above about disabling DMA for SD card.

kev018, 2017-05-02:
@mikeller Thx, i have done this yesterday but without a test flight. 
Is there any restriction? I do not find what the function causes.... 
Is there a Hard- or Software Issue?

mikeller, 2017-05-02:
The restriction is that logging will be slower, and as a consequence the maximum rate for logging without dropouts will be lower.
It's a resource conflict, and the setting changes SD card writes to not use DMA.

KenMurphy, 2017-05-16:
Just adding that I'm experiencing the same issue with an Omnibus F4 V3 Pro. Copter will fly perfect for x packs and then just randomly experience sudden extreme oscillations while shooting off in whatever direction it was facing at the time. 8k/8k looptime, 2K logging to SD card. Log seems to cut before the oscillations start (analyzing the log it happened in was a real head scratcher before I figured out that the log must have been truncated..).

Seeing that disabling DMA for the SD card might be a solution, I'm going to set that and fly some packs tonight to see if the issue crops up again. Will report back with my findings.

KenMurphy, 2017-05-17:
I set 'set sdcard_dma = off' and flew over 20 packs tonight without issues. Looks like that is a good temporary fix.

alessandro-montanari, 2017-05-24:
I had the same problem with an Omnibus F4 V3 (non Pro), here is the video https://www.youtube.com/watch?v=O9Sj2t8hNNU
I was running gyro and pid loop at 8k/8k and blackbox logging at 8k.
Now, I disabled the DMA for the SD card and lowered the logging rate to 2kHz. I flew three batteries and the quad didn't freak out but it seems that when I enable blackbox there are some vibrations, like excessive P gain, even with default PIDs. With blackbox disabled it flies great. Lowering the logging rate to 1kHz seems a bit better, less vibrations. Might be something to do with the SD card type? I am using a class 4 card.
I need to fly more and see how it goes.

gordonBimig, 2017-06-06:
Apparently this isn't a rare problem. I had a few crashes with my quad and destroyed a frame by inspecting the issue for the quad goind wild ...
Finally I figuered out, this behaviour only appears, when I turn Black Box logging in (8K PID Loop) and 2K looging. I have 4 Quads with almost identical setups and all others were ok, sao at first I assumed it's a faulty flight controller (all are BeeRotor F4 FCs).
So in my case the problem probably is beeing causes by a crappy 1GB SD Card which seems not to keep up with the write requests, although i thought 2Khz should be a to big deal, but aparrently it is.

I can confirm almost everything people before me wrote. Haven't tried the dma hack yet, but the SD Card will be replaced, so that there might be not need for this.

However, I'm wondering:
1. How come this issues hasn't been closed yet? I my opinion it is pretty severe.
2. How come, that a crappy SD card is able to cause such a behaviour when logging? Why isn't the FC simply dropping the packets?

I have a few videos showing the quad freak out, if it might helf and also black box file where the log gets interrupted and the quad freaks out but calms down again. If needed I can provide them, but the already posted links basically show the issue.

One also pretty severe thing I noticed with this issue is, that the quad doesn't diarm instantly when the quad freaks out. In my last case, which destroyed a frame, the quad was going crazy 2-3 secons AFTER I already hit the disarm switch, causing the quad to go full power upside down into the ground.

This was really a pretty scary experience, not beeing able to disarm the quad instantly.

KenMurphy, 2017-06-06:
> How come this issues hasn't been closed yet? I my opinion it is pretty severe.

I agree, at the very least all the affected boards need a big fat warning on their wiki page about this until a proper fix can be implemented. It's a pretty severe and dangerous bug. I got lucky that every time it happened to me (4-5 times until I found the temp fix here) the copter was facing in the opposite direction but it could have easily taken off towards me or somebody else because it happened so randomly with no specific flight behavior triggering it. 

I haven't had it happen in the last 20 days after setting sdcard_dma=off, but the behavior is pretty hard to google and find the fix for because of so many unrelated results.

hydra, 2017-06-06:
@mikeller what is the resource conflict with? DShot or something else?  Do we have a full list of boards affected?

mikeller, 2017-06-06:
@hydra: I don't know, honestly. I suspect it could be to do with DMA priorities, and the SD card DMA blocking DShot DMA requests. As far as I know, it only appears when DShot is used, and in some cases it goes away by disabling `sdcard_dma`, without the logging rate being reduced.

martinbudden, 2017-06-06:
Do we have `sdcard_dma` set off by default?

mikeller, 2017-06-06:
No. I suggested that a while ago, but @borisbstyle was opposed to it because the reduction in maximum possible logging throughput.

martinbudden, 2017-06-06:
Well, given that the problem doesn't seem to be infrequent we should revisit that decision. People can always change the value if they want.

hydra, 2017-06-06:
I think if we turned it off by default the problem would be forgotten and it'd never get fixed.

gordonBimig, 2017-06-07:
Hey guys,

in my case it is NOT DSHOT related. All my ESCs run  BLHeli_S and Multishot protocol. In my case it seems, betaflight is causing this behaviour because it's not able to deliver the blackbox log to the openlog device/sdcard as fast as it would like ... in my case caused by a crappy SD Card.

I'm not aware about the technical things going on here, but I think this MUST not happen, even if it is a 10 year old SD card (which cannot handle those fast many write requests), the packets "simply" should be dropped ... again no clue about whats going on here under the hood.

If it helps, I can provide a video/blackbox log showing you the quad is running Multishot and therefore it shouldn't be a DSHOT specific issue?

martinbudden, 2017-06-07:
@hydra , this https://blog.frankvh.com/2012/01/13/stm32f2xx-stm32f4xx-dma-maximum-transactions/ indicates that it might be a problem with the STM processors, in which case it cannot be fixed.

gordonBimig, 2017-06-07:
Just made a test today morning to confirm my assumption.

So I replaced the SD Card on the drone which was having the issue, with a SD Card from one of my others drones (which also has the BeeRotor F4 FC installe) and blackbox logging worked immediately without the quad freaking out. So to me the assumption, that it is SD Card related is confirmed and has nothing to do with DSHOT ... or at least not in my case?

martinbudden, 2017-06-07:
@gordonBimig , do you have `sdcard_dma` set to ON or OFF?

gordonBimig, 2017-06-07:
I haven't changed this value yet, so by default it should be turned on I guess. Betaflight 3.1.7

alessandro-montanari, 2017-06-07:
I had the same problem on Multishot as well.

gordonBimig, 2017-06-07:
@alessandro-montanari 
Have you tried another SD Card? The 16GB Transcend has been marked as good Blackbox Logging SD Card by Joshua Bardwell. I ordered one to try it out. On one of my other rigs I had this glitch for about second and it recovered itself ... so bad blocks or the sd card is on the edge of it's capabilities?

However, I still would like to understand what's going on here ;-)
How can a bad SD Card cause such dangerous trouble/behaviour when Blackbox Logging is ON?

alessandro-montanari, 2017-06-07:
@gordonBimig 
Before I was using a a Kingston 4GB class 4 card and I had the issue with Multishot and Dshot600. Then I turned the DMA for SD card off and changed the card (an anonymous 16GB I had laying around, I guess is Samsung) and I didn't have the problem since then. However I am running Dshot600 and I didn't try again Multishot with a different card and/or with DMA off.

KenMurphy, 2017-06-07:
@gordonBimig 
How many pack did you fly for that test? I've flown 10 packs in a row without issue and then it would randomly strike again. This is before turning off DMA of course.

Using a no-name unknown class 2GB SD card here as it was just laying around.

brycedjohnson, 2017-06-07:
I was getting the bug on a omnibusf4sd on a 3.2 build flipping out on an
arm randomly.  I had blackbox starting to log on arm.  Acted like a filter
was bad and shot straight up.  Changed the dma to off and haven't had an
issue yet (10-15 packs).

gordonBimig, 2017-06-07:
@KenMurphy
Just did 1 quick pack today morning before work. Will fly this quad more intensly on the weekend ... but ... my investigation shows ... it's a issue with the SD card ... so all "DMA" conviction aside ...

Can't anybody explain to me how this behaviour is possible? DMA On or Off ... doesn't matter. We have a scenario where the FC is not able to deliver the blackbox data to the SD card => how or why is the quad flipping out in this case??? Can't this be handled by simply dropping the packets WITHOUT the quad freaking out?

This is the question which is going through my mind all the time and nobody wants (or can) answer it ;-)

KenMurphy, 2017-06-07:
@gordonBimig 

It isn't purely an SD card issue though. Since turning off DMA on my build I haven't had the issue once. Using the same SD card as before.

brycedjohnson, 2017-06-07:
@gordonBimig 
It might be that keeping up to the SDcard isn't the problem or a bad SDcard.  From the website linked above: 

> DMA2  controller could corrupt data when managing AHB and APB2 peripherals in a concurrent way.

If it is related to that silicon bug getting a really fast sdcard won't fix the issue.

compudaze, 2017-06-07:
Happened to me on Omnibus F4 V2. 2k logging, SDCARDDMA was ON, brand new Samsung EVO 32GB UHS-1 card. Turned DMA off and reduced logging to 1K and haven't had issues since. Same SD card.

gordonBimig, 2017-06-08:
@KenMurphy 
OK. let's place bet's ;-)
I'm pretty sure it won't happen anymore, at least not as severe as I had it with the 1GB super old SD card.

The problem was easy replicable (as soon as I got i figured out). Replaced the SD Card with another one ... didn't touch any Betaflight settings and the quad flew fine (with blackbox logging on) IMMEDIATELY. 

Maybe this is just to high "math" for me, but in my opinion it's clearly a SD card issue and the SD card not able to keep up with the write requests.

We know Blackbox Logging is causing many write request in a short time and not neccessary high data throughput. We also know that a high end class 10 Micro SD card (optimized for fast data writing of large files) isn't neccessary the best choice for a Blackbox logging application (Video: https://www.google.de/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjDytjEqK3UAhXJcRQKHfhTDEkQtwIIIzAA&url=https%3A%2F%2Fwww.youtube.com%2Fwatch%3Fv%3DC8Cun7aWnVo&usg=AFQjCNFufgw_sMdr19ugqVlJkKoReIsjSg&sig2=DC9qHZVRaNkP7ori9hDAiA).

I think in all this problem scenarios, the used SD Card hits it's limit, which in every case can be a different one, depending on the SD Card and on the Blackbox logging parameters.

So what does the parameter SD Card DMA = Off do?

If I'm not wrong, it's diabling the FC to use a "DMA Channel" for SD Card writing => therefore it is freeing up ressources (which e.g. are used by DSHOT) and another "write method" is used, which is SLOWER and therefore setting this parameter to OFF, it can solve the problem in many cases by lowering the "write speed" (indirectly). I also read that people reduced the Logging ratio in most of this cases.

Just a theory :-D

gordonBimig, 2017-06-08:
@brycedjohnson 
I see, if it is a bug in the STM F4 Chip, when using DMA .. it would somehow explain why betaflight is freaking out, because the Chip might do stupid (unpredicted) things.

But like I said, on all my quads the SDCARDDMA parametere is default (so it is ON). 

> If it is related to that silicon bug getting a really fast sdcard won't fix the issue.

So if is related to this bug, I should have this issues on my other quads as well and also EVERY F4 board I assume?

brycedjohnson, 2017-06-08:
Mine didn't happen every time.  Maybe you have a separate different issue?  Mine would flip out on arm (and blackbox starting) every ~1/4 to 1/10 of the time.  If I arm and it doesn't flip out then there was no issue the whole flight.

Try it with DMA off and see if it fixes it for you.  It has for me. Not the root cause fix, but a workaround for now.

gordonBimig, 2017-06-08:
> Mine didn't happen every time. Maybe you have a separate different issue? Mine would flip out on arm 
> (and blackbox starting) every ~1/4 to 1/10 of the time. If I arm and it doesn't flip out then there was no 
> issue the whole flight.

I assume you don't have such a bad SD card like me ;-)
In my case I could replicated it 100%. Exactly like you said. If Blackbox logging was on while arming, the quad immediately started freaking out, without even touching the throttle. Another interestig fact might be, that this is only happening, when the props are ON!!! I tried to replicated this with props OFF and this was not possible, but each time the props were on, it was replicable.

Also when inflight blackbox logging was activated, it didn't take a second for the problem to appear. So really 100% reproducable. So to the beaflight developers, your welcome to try this, just find a really crappy SD card ;-)

If it's a STM F4 Chip bug, it might explain, why the quad didn't disarm immediately, because the chip was freaking out .. however, if this is the case ... SDCARDDMA should be set to OFF on all F4 board ... for safety reason in my opinion.

KenMurphy, 2017-06-08:
@gordonBimig 
No bet necessary, just please be careful :) I'm more or less just saying I wouldn't trust the new SD card being a perfect fix. I do think you're right in that the SD card speed does effect when the bug happens. But given that mine acts totally normal 95% of the time and then randomly freaks out I wouldn't put 100% faith in just swapping the SD card and calling it good, especially when disabling DMA appears to be a 100% workaround with no downsides.

gordonBimig, 2017-06-09:
@KenMurphy 
I'm afraid I have to destroy your workaround theory :-(

I've been doing a lot of testing and blackbox logging/analysis, because I'm "trying" to find perfect PIDs for my quads.

Today morning i flew 8 packs. Since I was a little bit tired fiddling around with the quad freaking out issue in certain scenarios, I followed your advice to put the paramater sdcard_dma to "off", on all quads while asuming, I'm finished with this. Unfortunately I'm not completely (yet).

I reported that there is one quad (or rather one SD card, since it is SD card related) which was giving me short glitches (1-2 seconds) quite rarely, but they exist and fortunately the quad didn't freak  out that much, that it results in a crash.
Since I now only have 3 SD cards for blackbox logging (for my 4 quads), because the crappy 1GB has been destroyed and the ordered 16GB Transcend hasn't arrived yet, I need to swap the SD cards arround to do blackbox logging and now I can tell, that all this (little) glitches which I had,  happened if the quad was logging on my 4GB Transcend (class 4 I assume, not written on) card (which also is quite old and probably not the best for this application)! So again there is this coincidence for the glitches to the SD card. The blackbox log clearly shows a break, while the quad freaked out.

So this is nothing new, right? Well in the last incident I had the parameter sdcard_dma set to "off" ....

So apparently this isn't bulletproof either. Unfortunately the blackbox log doesn't show all parameters of betaflight or at least not in the Chrome App GUI. Would love to prove that to you.

So this substantiates that this is more likely caused by bad/slow SD cards (or to high logging parameters/data), rather then the STM F4 DMA bug. Just my experience so far. I still believe that I'll be able to eliminate the issue with a really good SD card. Will keep you up2date if this belief doesn't hold up ;-)

[LOG00003_dma_off_motorsoftmount_Gopro_and_glitch.zip](https://github.com/betaflight/betaflight/files/1063371/LOG00003_dma_off_motorsoftmount_Gopro_and_glitch.zip)

ennead000, 2017-06-12:
Having the same issue on a Betaflight F3 AIO board, will try sdcard_dma = off and lowering logging rate to 2k tomorrow.  Was flying perfectly fine before inserting sdcard for notch filter debugging... so it's not just F4 fc's that are having issues.

Dutchy-79, 2017-07-15:
Also had a weird problem
Hex built with Matek F405-AIO and blackbox @ 1khz
Arming and flying fine, Then disarm and only a few motors spin and the others twitch

Turning DMA Off fixed it.

zhangyu0615, 2017-08-05:
i made a test, most of u can replicate my situation. Weather I set sdcard_dma on or off, when the quad armed, I plug the sd card in and out, to simulate a bad contact, the motor will twitch a little bit everytime. If with props on, the quad must had a attitude change, and then get into crazy viberate

gordonBimig, 2017-08-13:
Is this still relevant in Betaflight 3.2.0 RC3?

Has this been addressed?

Should I still use sdcard_dma = off?

mikeller, 2017-08-13:
@gordonBimig: This is still relevant, and if you had read this comment above https://github.com/betaflight/betaflight/issues/2631#issuecomment-306694919, you would know that this is potentially not fixable.

fiveangle, 2017-09-06:
Regarding @mikeller's statement above about @martinbudden's comment regarding the F4 errata:

[From: [STM32F40x and STM32F41x Errata sheet - July 2017 - DocID022183 Rev 9](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/0a/98/58/84/86/b6/47/a2/DM00037591.pdf/files/DM00037591.pdf/jcr:content/translations/en.DM00037591.pdf)]:

> **2.1.10 DMA2 data corruption when managing AHB and APB peripherals in a concurrent way**
> 
> **_Description_**
>
> When the DMA2 is managing AHB Peripherals (read- or write-sensitive devices such as peripherals embedding FIFOs or GPIOs) and also APB transfers in a concurrent way, this generates a data corruption (multiple DMA access). When this condition occurs:
> • The data transferred by the DMA to the AHB peripherals could be corrupted in case of a FIFO target.
> • For memories, it will result in multiple access (not visible by the Software) and the data is not corrupted.
> • For the DCMI, a multiple unacknowledged request could be generated, which implies an unknown behavior of the DMA.
>
> AHB peripherals embedding FIFO are DCMI, CRYPTO, and HASH. Also we can consider external FIFO controlled by the FSMC and GPIO Output Data register as AHB write sensitive peripherals.
> On sales types without CRYPTO, mainly impacted peripheral is the DCMI peripheral which embeds a FIFO. External FIFO controlled by the FSMC and GPIOs when used as parallel output are also impacted.
> 
> **_Workaround_**
>
> Avoid concurrent AHB (DCMI, CRYPTO, HASH, FSMC with external FIFO, or GPIOs output data register) and APB transfer management using the DMA2.

Looks like same is true in [F2 errata - 2.1.8](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/79/08/a3/cc/fa/5c/49/c5/DM00027213.pdf/files/DM00027213.pdf/jcr:content/translations/en.DM00027213.pdf), but the [F1](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf), and [F7](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/4d/39/a6/8c/84/47/47/67/DM00145382.pdf/files/DM00145382.pdf/jcr:content/translations/en.DM00145382.pdf) do not appear to have the issue.  F3 errata sheets seem to have been purged from the Interwebs as far as I can tell (perhaps they now reside behind a registration wall ?)

DieHertz, 2017-09-06:
F4 is a reiteration of F2, while F1, F3 and F7 are largely unrelated. Still it doesn't sound like the issue we have here, does it?

fiveangle, 2017-09-06:
> F4 is a reiteration of F2, while F1, F3 and F7 are largely unrelated. Still it doesn't sound like the issue we have here, does it?

F3 is _very_ closely related to F1.  F4 is similarly related to F2 (but not F1), so I agree with you that it may not affect F3, but design completion went F1 (M3) 2008, F2 (M3) 2010, F4 (M4F) 2011, and F3 (M4F) 2012 so it's a crap shoot really.

If I was a betting man I'd guess problem lies in the F2 peripheral interface design, so may have not have affected F3 since it likely kept similar F1 peripherial design, even though it hit the streets right after F4 (6 yrs after F1).  HW designers never reinvent their own wheel unless the market forces them to, and F3 was really just a refresh of the aging F1 arch with the much more capable Cortex-M4F dropped in the middle.

I'm sustaining engineering crises mgmt and not SW eng so I can't immediately tell if this errata is related or not, but this part gives me pause: 
> External FIFO controlled by the FSMC and GPIOs when used as parallel output are also impacted.

DieHertz, 2017-09-06:
I meant that F1, F3 and F7 aren't related to F2/F4. F2 had this issue, F4 inherited it, other chips likely don't have it.

C0br4, 2017-12-14:
As I mentioned in issue #3717 I have this problem even with sdcard_dma turned off.

[1 more reply]

closed 2018-04-21 by DieHertz