betaflight/betaflight #2633: flash writing broken since 32a6409fc4fe7bd6a13f8157ee0eb306f822b0a0

hydra, 2017-03-12:
dominicclifton [12:18 PM] 
@martinbudden lo

[12:18]  
SPRacingF3NEO is broken in master, fails to write eeprom (edited)

[12:19]  
i bisected it, and this is the commit that breaks it

[12:19]  
1964362dce2794c43c0e55354937166904802fda

[12:23]  
https://github.com/betaflight/betaflight/commit/1964362dce2794c43c0e55354937166904802fda
GitHub
Removal of USE_PARAMETER_GROUPS 1 · betaflight/betaflight@1964362
Contribute to betaflight development by creating an account on GitHub.
 

dominicclifton [12:50 PM] 
am i missing something here?
flash writing of F3 is broken, other F1/F4/F7 not tested, problem doesn't show up unless the config is large enough to use more than one flash sector.

here's a discovery log:

https://github.com/betaflight/betaflight/blob/master/src/main/config/config_streamer.c#L34

on an EVO or NEO this will result in a FlashPageSize of 4096 / 0x1000

however, in this commit

https://github.com/betaflight/betaflight/commit/32a6409fc4fe7bd6a13f8157ee0eb306f822b0a0

we can clearly see that the flash page size for an F3 is 0x800

this line of code runs on an F3:

https://github.com/betaflight/betaflight/blob/master/src/main/config/config_streamer.c#L215
 
so what happens, is eventually the c->address goes onto the next block, which has not been erased yet.

the reason that the issue doesn't show up on all targets (or until a certain commit is used) is likely due to the size of the config structure being different between commits.

it fails here:

https://github.com/betaflight/betaflight/blob/master/src/main/config/config_streamer.c#L227

you get the -2 return value.

```
static uint32_t lastEraseAddress;
static uint32_t lastProgramAddress;

...

    if (c->address % FlashPageSize == 0) {
#if defined(STM32F4)
        const FLASH_Status status = FLASH_EraseSector(getFLASHSectorForEEPROM(), VoltageRange_3); //0x08080000 to 0x080A0000
#else
        const FLASH_Status status = FLASH_ErasePage(c->address);
#endif
        lastEraseAddress = c->address;
        if (status != FLASH_COMPLETE) {
            return -1;
        }
    }
    lastProgramAddress = c->address;
    const FLASH_Status status = FLASH_ProgramWord(c->address, value);
    if (status != FLASH_COMPLETE) {
        if (lastProgramAddress > lastEraseAddress + FlashPageSize) {
            return -3;
        }
        return -2;
    }
```

the above piece of code highlights the problem

when it fails (with -3) the values are

lastEraseAddress = 0x803f000
lastProgramAddress = 0x803f800

which is *exactly* the place where it should have erased the next block if the flash page size was not incorrect

the commit the broke it was this:
https://github.com/betaflight/betaflight/commit/32a6409fc4fe7bd6a13f8157ee0eb306f822b0a0

unitware, 2017-03-12:
solving this should solve this as well https://github.com/betaflight/betaflight/issues/2615

hydra, 2017-03-12:
@unitware i think it might be unrelated, since when it fails you get a hardfault when the flash is invalid or erased and the blinky light pattern.  If you don't get that behaviour in #2615 then it's probably not the solution to #2615

hydra, 2017-03-12:
I'll create a PR.

hydra, 2017-03-12:
@unitware with the #2634 applied, 'cli dump' does work for me on my F3 NEO if that helps.

unitware, 2017-03-12:
Thanks, I'll check it later tonight.

commit 0f192560 (2017-03-13): Merge pull request #2634 from betaflight/fix-flash-erase-1

CF/BF - Fix incorrect flash page size configuration.  Fixes #2633.
commit 633aa6cb (2017-03-13): CF/BF - Fix incorrect flash page size configuration.  Fixes #2633.

Broken by 32a6409fc4fe7bd6a13f8157ee0eb306f822b0a0
closed 2017-03-13 by martinbudden