betaflight/betaflight #2635: SD Card issue SPRacingF3EVO RevE & BF 3.1.6

UltiBots, 2017-03-12:
Hi All -

I have read #611 & #1742 and others. SD Card is not working with SPRacingF3EVO RevE & BF 3.1.6

```
# sd_info
SD card: Startup failed
```
Betaflight / SPRACINGF3EVO 3.1.6 Feb 21 2017 / 14:44:42 (1fd502c)
Not using DSHOT
Set sdcard_dma = off
Formatted with SDFormatter
Board is Rev E

[image]

mikeller, 2017-03-13:
Is it never working? In that case my guess would be bad hardware. Some people have reported being able to solve problems with bad contacts by sticking a piece of sellotape to the back of their SD card before inserting it into the slot.

UltiBots, 2017-03-13:
The SD card never got passed "startup failed". The contacts should be good as the board is new. I will try tape as you suggest to get better contact.

TheCluster, 2017-03-13:
I have two sp racing f3 evo rev D. On both setups the sd card is working (sdcard_dma = off) with dshot and multishot. BF 3.1.6

UltiBots, 2017-03-13:
Issue resolved... Must enable SD Card in addition to Blackbox. Was using an older video tutorial that skipped this step - Doh!

closed 2017-03-13 by UltiBots