betaflight/betaflight #2637: Serial rx IBUS does not work on master

unitware, 2017-03-12:
When running master (@32a6409fc4fe7bd6a13f8157ee0eb306f822b0a0 - Merge pull request #2630 from mikeller/fixed_cli_get_set) on an SPRACINGF3 and selecting serial rx, IBUS the receiver channels have strange values. 

Pitch, Roll, Yaw and throttle are all 1000.
Aux 1,2,3,4 are all connected and responds simultaneously to roll input from the transceiver.

```
CLI
# diff
# Betaflight / SPRACINGF3 3.2.0 Mar 12 2017 / 18:56:44 (633aa6c)

feature -RX_PPM
feature RX_SERIAL
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
set serialrx_provider = IBUS
profile 0

rateprofile 0
```

```
f0b746107560f0893b0a79f897c8b5da222df0e6 Merge pull request #2636 from martinbudden/bf_scheduler_tidy
463ee8af2d9f0fbe246bae242dfa422c5e1a99ac Minor scheduler tidy
207bfd43e3d48116dce42163eff7a2c8b367aa75 Merge pull request #2630 from mikeller/fixed_cli_get_set
441b0c4e9836adf2734f0a3db78a7459054eaf63 Fixed 'get' / 'set' CLI commands.
454081978b963ee2865d2c949064e8412aef6e3d Merge pull request #2627 from GTorreil/patch-1
...
```

unitware, 2017-03-13:
This issue remains, look at this video to see the coupled aux 1-4, never mind the changing values on pitch, roll, yaw and throttle. They are moving only because of the failsafe, no stick movement will alter them.

Aux 5 and 6 is working though.

https://www.dropbox.com/s/iqsktknz2zu0a67/2017-03-13%2021.32.33.mov?dl=0
(sorry about the dropbox link, the file was to big)

unitware, 2017-03-14:
I have been test building a lot of revisions, last known version that worked is 90443bb (thats HEAD~17 right now) after that the settings does not work so I cannot test.

McGiverGim, 2017-03-21:
I can confirm the issue, in my NAZE32. I can't use my FlySky system with head revision.

atomiclama, 2017-03-22:
Was going to look into this but just tested on a CC3D with an IA6B

Master with cheery pick #2725 

It is showing up correctly in the configurator receiver tab.

Sorry can't test on a Naze32
same diff as above but with 

`
resource SERIAL_RX 5 D00
`
Perhaps this is missing from your setup, maybe perform default after the update or a full flash erase.
I'm programming via ST-LINK so a bit different from the norm.

I'll try on an AIRBOTF4 clone when I get a chance.

@unitware do you still need some traces for the iBus protocol? Been a bit busy and only just getting back into this.

unitware, 2017-03-22:
Hi, I don't need the traces, made my own. And thanks for the report, will try it again tonight. There has been a lot of commits in the vicinity of rx serial and parameter groups lately that might have addressed it.

McGiverGim, 2017-03-23:
Hi @unitware ! Did you try again this? 

One more thing: I've seen in your code and documentation that you reference the fs-ia6 and fs-ia6b. Is there any problem if I use the fs-ia10b? In your cleanflight PR it worked perfectly, I don't know if something has changed in your betaflight PR.

Regards!

unitware, 2017-03-23:
Hi, did not have time to test yesterday, sorry. But I heard that another user managed to fly yesterday but had no yaw, so it might hav gotten better but not good.

There should be no change regarding fs-ia20b as far as I know. the change is addition of fs-ia6 which uses a slightly different format.

McGiverGim, 2017-03-23:
I've the quad broken, but at least in cleanflight, with only the receiver connected to the FC now it's working, yaw included.

When the repair parts arrive, I will test one by one to see if it's solved or some part conflicts.

McGiverGim, 2017-03-23:
Maybe the yaw problem is related to: https://github.com/betaflight/betaflight/pull/2684

It was merged 5 days ago.

unitware, 2017-03-23:
tested ok today. The problem has been solved prior to e4020d9e9955a11c613fe6289be7bdbfc416ae09

Test on SPRACINGF3NEO, everything seemed fine pitch, roll, throttle, yaw and aux 1-6.
I did have another problem though - with blackbox logging to the SD-card which was not recognized.

closed 2017-03-23 by unitware

McGiverGim, 2017-03-23:
I had a similar problem: now in the blackbox tab you can select embedded or serial. Select embedded.

unitware, 2017-03-23:
Thanks, I'll dig deeper in that in a few days - limited quad time ahead I'm afraid.

unitware, 2017-03-24:
Blackbox on sdcard problem resolved: https://github.com/betaflight/betaflight/pull/2740