betaflight/betaflight #2638: Motor spin out of control

fourknock, 2017-03-12:
Halo i have some isssue
I try to running dshot 600 at emax femto f3 with racestar 6a v2 2-3s dshot support(with put signal caps)
I use led pads and remapping from cli
Now all the motor is running but 
When start arming and increses trothel speed/rpm motor not follow my stick it keep increse to max
I already flash BF with difrent firmwer and also esc firmwer,change new board 
When i do bench test motor running smooth with out increse to max stop base on value i input at BF
Thanks i hope i get solution about mu problem

mikeller, 2017-03-12:
Could be an air mode / configuration problem. Can you post a config `diff`?

jflyper, 2017-03-13:
Isn't this just another case of spool up on bench?

fourknock, 2017-03-13:
https://youtu.be/0yBcE-B7C4k 
I make short video of the issue

fourknock, 2017-03-13:
# diff
# Betaflight / SPRACINGF3EVO 3.1.5 Feb  7 2017 / 22:31:13 (4646f9d)
resource MOTOR 4 A08
resource LED_STRIP 1 NONE

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -BLACKBOX
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
serial 2 64 115200 57600 0 115200
aux 0 0 0 1700 2100
set motor_pwm_protocol = DSHOT300
set serialrx_provider = SBUS
profile 0

rateprofile 0
rateprofile 0

mikeller, 2017-03-13:
Yes, this is just bench spinup without props on. The flight controller tries to compensate for the input it gets from the gyro but obviously can not do so since there are no props, so it spins up...

fourknock, 2017-03-14:
I already try fly,kwad fly straight up and i can control or maintain altitude........i can only disarm to stop kwad fly

darkusone, 2017-03-14:
It happens the same to me and it has me crazy ... 
I have bought 3 FC, I explain myself:
now I have 2 quads one worked perfectly and I decided to change for a better frame and other engines and ESC, well, I start to build the new one and when I mount my second fc and I'm going to try it I see that it "falls" to one side, As if we were in airmode, I corrected the pitch, a little roll and landed, I checked the flight modes, and I was actually in ANGLE mode, I take the GUI to check the accelerometer, everything seems OK, I remove props and I simulate to fly That the ESC respond correctly, and I see that it happens to me just what the creator of the thread happens to him, it seems that he is trying to correct a nonexistent inclination, and that really in the 3D model does not have!!! (is leveled).

Then I start to do tests and as I have the previous drone and it worked well, change the fc to see if the same happens, flash, set everything up, and FOLLOW the same !! At the moment I just changed all the following in case any component is wrong:
Radio receiver
- frame (the other is carbon fiber)
- ESCs
- Engines
- 3 different fc (new)!!
- battery
- PCB + BEC

Initially I thought it would be the accelerometer that would have broken "casually" on the 2 plates and buy another yesterday, and nothing ... the same ...

I only have to try the remote control, nothing more, but it has battery to the maximum, I am to 2 meters and the graphs of frequency everything is smooth when I do not touch anything ...

I do not know what to do anymore

jflyper, 2017-03-15:
Try 3.1.6.

Just a vague memory, but we had motors freaking out on a certain patch release of 3.1, which has been fixed.

darkusone, 2017-03-15:
yesterday, after post, i was watching viedos, reading documentation like FAQ (https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-do-my-motors-keep-accelerating-on-the-bench-when-i-arm-without-props-) 

"##Why do my motors keep accelerating on the bench when I arm without props ? With props off on the bench, I arm the quad and the motors start. After increasing throttle a small amount then back to minimum I notice the motors keep increasing in speed. They don't go to max or anything, but they climb noticeably. Now if I was in Angle/Horizon with the accelerometer enabled I could understand that the quad was tying to level itself. But in Acro mode why should the throttle change on its own ? I'm guessing this is an Airmode effect. But just wanted to understand a little more about why.

Answer: That is the flight controller trying to correct for changes in aspect, mainly due to fact your quad shakes slightly when the motors spin, the sensors pick it up and then the flight controller tries to correct, it can't because you don't have props on. All perfectly normal."

and more testing, etc, etc, etc, i concluded that may be a bad feeling, today afterwork i going to my flight area to really test if all is well, then i post again to solve future comments on this thread.

Thanks

fourknock, 2017-03-15:
Thanks for replay all,....it is not my first time build a multirotor before i post this issue i already test with prop....actualy test flight ,kwad  cannot hover,kwad not follow my stick input (trothel ) ...kwad fly straight up,when i try to reduce my trothel input kwad cannot dollow my stick input only when i hit zero trothel kwad cut off and i dont use airmode

darkusone, 2017-03-15:
Hello again
I've been flying, I've come back, and the quad is still alive !! Jejeje

Finally if, my quad is well, despite some necessary PID retouching works everything correctly.

Possibly @fourknock have a different problem, we could watch a video keeping the drone about 2 meters high and without touching pich and roll to see how it behaves, according to what I have been able to verify and what the documentation says, it is normal that the ESC (airmode included), no props and in the GUI go crazy.

jflyper, 2017-03-16:
@fourknock What is your version number? Should at least 3.1.4.

fourknock, 2017-03-16:
I use 3.1.5 in betaflight and bl heli 16.6
Soon i will post video with prop
My bye it will help to now what issue i have and solusion for that......

BatutaBR, 2017-03-17:
I think this is not a problem, but the normal operation of the AIRMODE function. Do a test, disable the AIRMODE function, and see if the problem disappears. If the problem no longer occurs, re-enable the AIRMODE function and you can fly. Basically with the drone on the table, without propellers, and after that you accelerate above 1350 (I think) active the AIRMODE, but its tries to correct the drone's position on the table and can not get, because its has no propellers. This gradually increases the rotation of the motors. Put the propellers, hold firmly the drone accelerate a little and you will activate the AIRMODE, the rotation will rise gradually, lower the throtle stick and the acceleration should reduce to idle gradually. This is the normal operation of the AIRMODE function.

darkusone, 2017-03-17:
@BatutaBR he says is not using airmode

johnnyruz, 2017-03-17:
@fourknock what is your min_throttle, or if you're running DSHOT, your digital_idle_percent set to?

fourknock, 2017-03-17:
@Batutabr ...no airmode 
@jhonyruz.  1070 min trothel
Digital idle 1500 and i already change the value it stil same

yyybap2, 2017-11-12:
I bought an SPC maker 95gf mini quad and had the exact same problem...  I armed the quad and all the motors would spin to maximum rpm even though I gave it a little throttle.   I tried many things before I was able to fly the drone.   If you go to the PID Tuning settings in beta flight...  there's a tab called filtering...  disable all 3 filters...  I found a youtube video that explains the motors will run hot with the filters turned on...  I also lowered the Angle and Horizon strength on the PID Tuning screen from 50 down to 30.   I saw a video where he lowered his angle to 10.  Another change, which I found in another youtube video... I change the Throttle Mid to .10 and the Throttle Expo to .70 on the PID tunning screen.  Turning the filtering off was the last thing I did and believe it was the key to stabilizing the motors.
Hope this helps...

r15i, 2020-11-26:
hi i have a similar problem with matek f405-std board, it goes perfectly fine testing the motor signals from the gui,  but when i arm and raise the throttle from the radio the flight controller raise the motors signal in a similar way to @fourknock and also if i full throttle , after half stick , i loose control and it just stay stuck at full throttle . i'm rocking the latest betaflight firmware for the board (4.x.x), gyro is in the correct orientation , matek specifies it quite clearly in the product page, not even putted pops on once, it

jflyper, 2020-11-27:
@r15i Are you experiencing the issue while you are bench testing? If so, it is perfectly normal.

And don't reply to almost 4 year old thread...