betaflight/betaflight #2640: Freeze when SmartAudio on SoftSerial (RMRC Dodo/SPRacingF3)

flounderscore, 2017-03-12:
I'm setting up a new quad and running into issues when using SmartAudio via SoftSerial.
Initially everything works fine but then after 10s of seconds the system freezes. I'm running BetaFlight 3.1.6 on a RMRC Dodo (RMDO or SPRACINGF3 firmware), MSP on UART1 (USB), UART2 (MinimOSD), and serial RX on UART3. SmartAudio connected to the TX pin of Softserial2 (1 doesn't work either).

> System Uptime: 32 seconds
> Voltage: 103 * 0.1V (3S battery - WARNING)
> CPU Clock=72MHz, GYRO=MPU6050
> Stack size: 2048, Stack address: 0x10002000
> I2C Errors: 0, config size: 1628
> CPU:10%, cycle time: 253, GYRO rate: 3952, RX rate: 91, System rate: 9

Initially "tasks" looks the following:

> 00 - (       SYSTEM)      9       5       1    0.5%    0.5%         0
> 01 - (          PID)   2000     176     145   35.7%   29.5%     14545
>    - (         GYRO)   4000
> 04 - (           RX)     90      62      52    1.0%    0.9%       128
> 05 - (       SERIAL)     98     263       3    3.0%    0.5%        15
> 07 - (      BATTERY)     49      13       6    0.5%    0.5%         6
> 08 - (       BEEPER)     99      10       2    0.5%    0.5%         5
> 17 - (          CMS)     59       6       3    0.5%    0.5%         3
> 18 - (      VTXCTRL)      4      85      63    0.5%    0.5%         5
> RX Check Function                17      10                        24
> Total (excluding SERIAL)                      39.2%   32.9%

and right before the system gets unresponsive (415.8%!):

> 00 - (       SYSTEM)      9       5       1    0.5%    0.5%         0
> 01 - (          PID)   1992     180     144   36.3%   29.1%     18644
>    - (         GYRO)   3984
> 04 - (           RX)     89      62      52    1.0%    0.9%       162
> 05 - (       SERIAL)     99   41957       3  415.8%    0.5%        60
> 07 - (      BATTERY)     49      13       5    0.5%    0.5%         8
> 08 - (       BEEPER)     99      10       2    0.5%    0.5%         6
> 17 - (          CMS)     59       7       2    0.5%    0.5%         4
> 18 - (      VTXCTRL)      4      85      69    0.5%    0.5%         8
> RX Check Function                17      10                        31
> Total (excluding SERIAL)                      39.8%   32.5%

And my diff:
> feature -RX_PPM
> feature -TELEMETRY
> feature -CURRENT_METER
> feature -RSSI_ADC
> feature -TRANSPONDER
> feature RX_SERIAL
> feature SOFTSERIAL
> feature AIRMODE
> map TAER1234
> serial 1 1 115200 57600 0 115200
> serial 2 64 115200 57600 0 115200
> serial 31 2048 115200 57600 0 115200

jflyper, 2017-03-13:
@flounderscore Do you know wether your SmartAudio device is V1 or V2?

flounderscore, 2017-03-13:
It's a Unify Pro V2.

jflyper, 2017-03-13:
Funny... the SERIAL task has very little to do with SmartAudio processing; VTXCTRL task is...

Oh wait. Which pin did you assign your SOFTSERIAL1 TX (resource `serial_tx 11`) to?

flounderscore, 2017-03-13:
Currently I’m using softserial2 (after first trying softserial1). I did not change any of the default pin assignments:

resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00

flounderscore, 2017-03-17:
@jflyper Any more ideas what I could try to debug my issue?

flounderscore, 2017-04-01:
I just verified. SoftSerial enable but SmartAudio not in use does not freeze my board.

jflyper, 2017-04-03:
@florian Can't reproduce the problem here.

- BF 3.1.6 (stock)
- RMDO Rev.1
- Futaba R7008SB on UART3 (Serial RX)
- MicroOSD 2.5 (MWOSD) on UART2 (MSP)
- USB on UART1 (MSP)
- Unify Pro 5G8 (SmartAudio V2) on SOFTSERIAL2 (RC_CH8/PB1)

[image]

```
# diff
# Betaflight / RMDO 3.1.6 Feb 21 2017 / 14:40:38 (1fd502c)

feature -RX_PPM
feature RX_SERIAL
feature SOFTSERIAL
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 31 2048 115200 57600 0 115200
set serialrx_provider = SBUS
profile 0

rateprofile 0
rateprofile 0
```

```
# status
System Uptime: 17 seconds
Voltage: 58 * 0.1V (2S battery - CRITICAL)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 1628
CPU:7%, cycle time: 506, GYRO rate: 1976, RX rate: 49, System rate: 9
```

```
# tasks
00 - (       SYSTEM)      9       4       1    0.5%    0.5%         0
01 - (          PID)    984     170     144   17.2%   14.6%      3607
   - (         GYRO)   1968
02 - (        ACCEL)    993     120     112   12.4%   11.6%      1505
03 - (     ATTITUDE)     97      40      34    0.8%    0.8%        45
04 - (           RX)     49      98      84    0.9%    0.9%        60
05 - (       SERIAL)     97     478       3    5.1%    0.5%        91
07 - (      BATTERY)     49      17       8    0.5%    0.5%         4
08 - (       BEEPER)     99      12       2    0.6%    0.5%         2
13 - (    TELEMETRY)    248      11       4    0.7%    0.5%        20
16 - (          CMS)     59       7       2    0.5%    0.5%         1
17 - (      VTXCTRL)      4      96      73    0.5%    0.5%         5
RX Check Function                 8       5                         3
Total (excluding SERIAL)                      34.6%   30.9%
```

flounderscore, 2017-04-08:
Thanks @jflyper for testing! I've continued with the build and it seems to be ok now. (the MWOSD in Canvas mode seems to freeze every now and then, but that is unrelated I believe). I did still have an issue while connected to the configurator, but oddly could not reproduce it (although I didn't try too hard).
I will flight test tomorrow and close the issue if everything is working.

jflyper, 2017-04-09:
@flounderscore Looking forward to hearing the flight test.

flounderscore, 2017-04-09:
The OSD menu was working nicely on using canvas mode on MWOSD and TBS unify on softserial (BF 3.1.7 now, running SPFRACING firmware on RMRC Dodo). Not sure what caused my issues earlier. Maybe it only happened when connected to the configurator using USB? Either way, magically seems to be ok now. Thanks for your help @jflyper .

closed 2017-04-09 by flounderscore

t2000, 2017-06-16:
@flounderscore I am also trying to configure my SPRACINGF3 with TBS smartaudio and would like to attach my VTX to the softserial2 port. I have now soldered it to pin6 of the I/O2 which is next to the beeper pins.

If I enable TBS smartaudio on the ports tab for softserial 2 my telemetry stops working completely :(

With this config telemetry is working but smartaudio not configured:
```
diff
# Betaflight / SPRACINGF3 3.1.7 Apr  3 2017 / 22:27:23 (e1c4b5c)
name T2000-Racer

feature -CURRENT_METER
feature -RSSI_ADC
feature INFLIGHT_ACC_CAL
feature SOFTSERIAL
feature AIRMODE
serial 1 2 115200 38400 0 115200
serial 2 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
aux 0 0 0 1900 2100
aux 1 1 1 1900 2100
aux 2 2 1 1400 1600
aux 3 13 3 1900 2100
aux 4 26 4 1900 2100
set rssi_channel = 10
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 16000
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set serialrx_provider = SBUS
set align_board_yaw = 90
set gyro_sync_denom = 2
set failsafe_throttle_low_delay = 20
set pid_process_denom = 1
set blackbox_rate_denom = 5
profile 0

rateprofile 0
rateprofile 0
```

If I change the softserial2 port, i.e. enable TBS smartaudio there:

```
diff
# Betaflight / SPRACINGF3 3.1.7 Apr  3 2017 / 22:27:23 (e1c4b5c)
name T2000-Racer

feature -CURRENT_METER
feature -RSSI_ADC
feature INFLIGHT_ACC_CAL
feature SOFTSERIAL
feature AIRMODE
serial 1 2 115200 38400 0 115200
serial 2 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
serial 31 2048 115200 57600 0 115200
aux 0 0 0 1900 2100
aux 1 1 1 1900 2100
aux 2 2 1 1400 1600
aux 3 13 3 1900 2100
aux 4 26 4 1900 2100
set rssi_channel = 10
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 16000
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set serialrx_provider = SBUS
set align_board_yaw = 90
set gyro_sync_denom = 2
set failsafe_throttle_low_delay = 20
set pid_process_denom = 1
set blackbox_rate_denom = 5
profile 0

rateprofile 0
rateprofile 0
```

Telemetry stops working.

I have also tried to solder both tx and rx pins of the softserial to the microphone pin of the vtx as I needed this for softserial1 on the smartport of my frsky fr4rsb but that did not change anything.

What am I doing wrong?