betaflight/betaflight #2641: ibat_scale is reset when a 'diff' or 'dump' is performed in the CLI.

dingrgh, 2017-03-13:
If ibat_scale is changed either in the GUI or CLI (yes, and save+reboot after the change), the value is retained. But if a 'dump' or 'diff' is performed, it will revert to the default value of 220.

ibat_offset is not affected. Only ibat_scale.

lunddahl, 2017-03-17:
Something is fishy with ibat_scale, after flashing with Full Chip Erase or a Reset Settings, diff shows this:

```
# diff
# Betaflight / BETAFLIGHTF3 3.1.6 Feb 21 2017 / 14:14:49 (1fd502c)

set ibat_scale = 220
profile 0

rateprofile 0
rateprofile 0
```

dingrgh, 2017-03-19:
The default ibat_scale value for other targets is 400. I believe that the bug was introduced when Boris forced the default for the Betaflight F3 board to 220. That's why it only affects this specific target.

toastedcornflakes, 2017-03-29:
Not 100% sure it's linked, but [this](https://github.com/betaflight/betaflight/blob/13da4de7749714d1025fab7beedb653594c0d231/src/main/fc/cli.c#L1059) seems fishy: the stack could get tramped by a sufficiently long (in the "number of digits" sense) config value

AndersHoglund, 2017-03-29:
Sorry, I have missed this report completely. 
This problem is fixed by PR #2709 and is available in 3.1.7 alpha builds http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Maint/

dingrgh, 2017-04-04:
Fixed in 3.1.7.

closed 2017-04-04 by dingrgh