betaflight/betaflight #2646: Feature Request: Full two way Telemetry on Spektrum SRXL SPM4649T OSD

macht-nix, 2017-03-13:
first i have to say thank you for the time time you put in this...

"Volts"  works great i could remove the V Batt wires from my Receiver :wink:
i use OMNIBUSF4SD 3.1.6; TX on UART6 (SRXL & Telemetry on / Telemetry output disabled); DX9 1.11 EU-328

are there plans to implement more features like (most interesting for me):
- Flight log data (for me just Fades & Holds; others may like RSSI) to OSD with a warning in case of eg. more than 10 Fades/sec or Holds
- Flight Pack mah drawn to radio
- Altitude to radio
- some may like Current draw to radio

estevanb, 2017-05-02:
Hi Macht-nix. Can you detail your wiring? I am putting together a OmibusF4SD with SPM4649T and I am curious about it. Did you wire the bidireccional signal wire from RX direct to the TX pin of UART6 of the Omnibus board?

Thanks!

macht-nix, 2017-05-03:
Hi @estevanb ,

Yes the wiring was signal to TX-pin (UART6), ground to ground (UART6) and Power to 5V (UART6).

But since last week my SPM4649T is gone. My main ground wire (Omnibus) came undone in crash and ground looped back over the ESC signal wire. So maybe that's a thing to keep in mind. 
Now i removed the ground from the signal wire of my ESC and use a R110XL untill spektrum-telemetry gets implemented.... 60€ vs 12€ without advantages is just too much. 

On a sat RX i just solder RSSI to the blinky LED and i get RSSI (as % lost frames) into my OSD....

estevanb, 2017-05-29:
Thanks for the reply! I did it and it worked nice! Since I am using the SPM4649T, I am not getting RSSI info on OSD... Just the telemetry Flight Log on TX... (frame loss and holds.) Do you know if this technique of solder RSSI to the blinky LED would work for this RX instead of the SAT RX?

Thanks again macht-nix!

macht-nix, 2017-05-29:
I never tested it, but just watch the led …. if it blinks faster the more frame losses / time you get, is off in holds and gets back to solid when you get a good signal again. It should be worth a try.

…..but first test it with a multimeter, than try to solder it and be careful the soldering will not be easy its very small.

Abancalari, 2017-06-26:
Maybe also a he Vtx channel selection menu that comes on spektrum radios to the smart audio handler?  It's been implemented on RF for a bit now and seems to work flawlessly. Useful for people who don't have OSD

taileron, 2018-01-20:
I got it managed as well to get battery telemetry from the fc with SPM4649T. With gps working on F3 (currently only by removing other unused code components to fit into memory). 
It would be great to have information about distance to home with distance alarm from the tx and probably gps / baro details like altitude, time, groundspeed and position of dive into the vegetables for easier copter searching when flying without fpv/osd. Currently the fc sends only telemetry about the battery back to tx. All that telemetry data could be logged on the tx´s sd for later investigation e.g. if the copter gets lost.