betaflight/betaflight #2648: Betaflight 3.1.1 Motors Tab

HerrTim, 2017-03-13:
Hi,
I'm not yet very common with Betaflight but I wanted to test the vibrations on my motor. I've noticed that the Accelerometer has been replaced with Gyroscope and the "Reset" is not working. Can I change back the Gyroscope to the Accelerometer?

Thank you!

mikeller, 2017-03-13:
Reset has been fixed in https://github.com/betaflight/betaflight-configurator/pull/464 . What is your reason for wanting the accelerometer graph instead of the gyroscope graph?

HerrTim, 2017-03-13:
Do I have to update the file manually or how does it work? Sorry I'm a complete noob :-X
Probably it's also complete stupid, but the accelerometer was kind of "quiet". The gyroscope is always moving even if the quad is standing still on a table next to me without doing anything.