betaflight/betaflight #2652: SP F3 flight controller and XSR reciever Telemetry? please help

Jdog-glossop, 2017-03-14:
Can anybody confirm how I should wire a SP F3 flight controller to XSR receiver? It's the telemetry I'm having trouble with!
There are two diagrams online 1 says io-connector and uart2 smartport and diagram 2 says all uart2 and the smartport to rxtx ? 
First taranis radio 📻
Cheers jamie
[image]
[image]

mjedlinski, 2017-03-14:
You should connect XSR SBUS to one of UART RX Pin and S.Port (telemerty) to other USART TX Pin.
You don't need to connect telemetry RX and TX pin.

Jdog-glossop, 2017-03-14:
mjedlinski thanks for the reply is that diagram 2 then?

mjedlinski, 2017-03-14:
Both diagrams are ok.
In first diagram SBUS is connected to third pin of IO_2 and when You use serial receiver it acts as UART3_Rx Pin. S.Port should be connected to UART2 (as in diagram) but put it only to Tx Pin (it's not required to short Rx and Tx Pin).

In second diagram Sbus is connected to Uart2 Rx so S.Port should be connected to UART3 (Tx Pin only).

Jdog-glossop, 2017-03-14:
mjedlinski Thank you for replying fella

closed 2018-04-21 by DieHertz