betaflight/betaflight #2654: ACC wont boot BF3.1.6 SOULF4

xhitm3n, 2017-03-14:
Hi all,
So i was changing PID´s and then change the Horizon Mode Strength from 50 to 20, then i tried to fly and escs would arm, then i checked in configurator and it wont boot, i flashed again and tried clean flash, im on Demon Soul F4, here is a video:
https://www.facebook.com/groups/betaflight/permalink/435036953512730/
is it posible do disable ACC in DFU or boot up anyother way? this board was expensive its really shame that it is bricked now :(
what can i do? please help!
Thank you for your work betaflight is great!

mikeller, 2017-03-14:
Have you tried reflashing with 'Full chip erase' checked? This will reset your config to default.

xhitm3n, 2017-03-14:
Yes i have tried that, even tried Raceflight with no luck, tried different computer still same issue...

mikeller, 2017-03-14:
Looks like your board has a hardware defect then.

xhitm3n, 2017-03-14:
i have tried again and it gets stuck at flashing with the bar filled...

AndersHoglund, 2017-03-14:
Don't use Full Chip Erase on F4 boards, does not work. Known issue since long. https://github.com/betaflight/betaflight-configurator/issues/200

xhitm3n, 2017-03-14:
I flashed again latest build still not working :|

DieHertz, 2017-03-16:
I have 3 SOULF4 and all of them work on latest master, something's not right with your board.

closed 2018-04-21 by DieHertz