betaflight/betaflight #2655: BF 3.2.0 - CLI diff & dump reset settings

TheAngularity, 2017-03-14:
using latest master `6303de7` and bfc 3.1.1 ... if i'm set any config and go to cli and use dump or diff the config will listed ... but when i go back in any config tabs now the default setting is back ... also in addition the dump and diff outputs are incomplete

mikeller, 2017-03-14:
Configuration is being worked on in 3.2.0. Can you list what is missing from diff / dump?

TheAngularity, 2017-03-14:
in dump output missing the `# resources` list and the `# map` feels wrong 

```
# map
map 4
```
should be after default
```
# map
map AETR1234
```

on diff output the `# name` is set in config but the output is empty and also `acc_lpf_hz` is not set but diff output prints `set acc_lpf_hz = 256`

mikeller, 2017-03-14:
Will look into this, thanks for the report.

DieHertz, 2017-03-16:
Working on it.

DieHertz, 2017-03-16:
@TheAngularity you may try to apply patch from #2664 manually as it's awaiting to be merged in.

TheAngularity, 2017-03-16:
looks good ... config is now preserved after diff & dump ... but the only thing missing #2663

DieHertz, 2017-03-16:
Ah, name, forgot about that. Although `acc_lpf_hz` is not reproduced for me.

DieHertz, 2017-03-16:
Fixed name in a separate PR. Looking at serial. Need more info about `acc_lpf_hz`, what exactly did you do? Tried Full Chip Erase?

TheAngularity, 2017-03-16:
ok here the `# serial` after `defaults` and `dump`

```
# serial
serial 20 1 115200 38400 0 115200
serial 0 1 115200 38400 0 115200
serial 2 0 115200 38400 0 115200
serial 5 0 115200 38400 0 115200
```

and now `# serial` after disabled uart1 as msp and enabled uart6 for serial rx ... not done at the baud rates

```
# serial
serial 20 1 115200 57600 0 115200
serial 0 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

```

DieHertz, 2017-03-16:
@TheAngularity GPS baudrate changed spuriously?

TheAngularity, 2017-03-16:
imho the default baudrate has always been 57600 and not 38400

DieHertz, 2017-03-16:
@mikeller @martinbudden @TheAngularity 
Indeed, *BetaFlight Configurator* has default GPS baudrate at *57600*, and before PG addition it was *57600* in firmware, but iNav has *38400* by default. Should we change back to *57600* or set the same value as iNav?

TheAngularity, 2017-03-16:
ok ... `# resource` now back in output again

DieHertz, 2017-03-16:
Found the issue with `acc_lpf_hz`, invalid `offsetoff`.

TheAngularity, 2017-03-16:
> Indeed, BetaFlight Configurator has default GPS baudrate at 57600, and before PG addition it was 57600 in firmware, but iNav has 38400 by default. Should we change back to 57600 or set the same value as iNav?

@DieHertz 
imho if the targed have usb vcp the uart1 as msp was disabled on default ... but its now activated again

DieHertz, 2017-03-16:
That seems to be done on purpose by @martinbudden

TheAngularity, 2017-03-16:
@mikeller @DieHertz 
output from `# resource` would not really fixed ... here a compare between 3.1.7 / 3.2.0

[image]

mikeller, 2017-03-16:
@DieHertz:
- `name` was in `diff` / `dump` since it got added;
- GPS baudrate seems to be pretty unimportant in Betaflight, I'd set it back to default to 57600, to minimise disruption for people updating their config;
- do you know why MSP on UART1 for VCP targets was enabled again? It was disabled because it makes it impossible to connect to a craft after resetting the config if a device connected on UART1 is sending data (would @martinbudden know?).

@TheAngularity:
- yes, there still is a problem with the resource config for serial ports, probably due to the ioTags not properly being initialised. As mentioned in #2664, this should be addressed in a separate PR.

DieHertz, 2017-03-16:
@mikeller Regarding MSP on UART1, seems like it's just missing `defined(USE_MSP_UART)`. I'll add it back.

TheAngularity, 2017-03-17:
@DieHertz 
nice ... so far everything looks good ... only the problem with `# resources` is open

> yes, there still is a problem with the resource config for serial ports, probably due to the ioTags not properly being initialised. As mentioned in #2664, this should be addressed in a separate PR.

DieHertz, 2017-03-17:
Let me know if you find something else.
Will look into resource management with PGs, although I'm not sure I'll be able to do better than @mikeller.

AndersHoglund, 2017-03-20:
I can see the same, or similar, problem on BFF3 with 3.1.7 , `diff `reseting some value. In my case the `ibat_:scale` parameter is reset to the target specific setting defined in `src/main/target/BETAFLIGHTF3/config.c`.  Not to the internal value 400  shown by `diff showdefaults`. All other settings are preserved, only this target specific default value is reset.
Any target specific default setting will probably have the same problem, on other targets.

    # get ibat_scale
    ibat_scale = 220
    Allowed range: -16000 - 16000
    
    # set ibat_scale =430
    ibat_scale set to 430
    # get ibat_scale
    ibat_scale = 430
    Allowed range: -16000 - 16000
    # save
    
    # get ibat_scale
    ibat_scale = 430
    Allowed range: -16000 - 16000
    
    # diff
    # Betaflight / BETAFLIGHTF3_MSPUART 3.1.7 Mar 16 2017 / 21:18:44 (5a9ae10)
    name DTI Spadger180
    
    feature -TELEMETRY
    feature LED_STRIP
    feature AIRMODE
    map TAER1234
    led 0 11,15:E:AI:0
    led 1 10,15:E:AI:0
    led 2 9,15::AT:0
    led 3 8,15::AT:0
    led 4 7,15::AT:0
    led 5 6,15::AT:0
    led 6 5,15:W:AI:0
    led 7 4,15:W:AI:0
    aux 0 0 1 1700 2100
    aux 1 1 2 1700 2100
    aux 2 2 2 1300 1700
    aux 3 12 0 1000 1500
    aux 4 25 4 1000 1300
    aux 5 18 3 1000 1300
    set motor_pwm_protocol = DSHOT300
    set serialrx_provider = SPEK2048
    set ibat_scale = 220
    set gyro_notch1_hz = 120
    set gyro_notch1_cut = 90
    set gyro_notch2_hz = 220
    set gyro_notch2_cut = 150
    set yaw_motor_direction = -1
    set acc_trim_pitch = -2
    set blackbox_rate_denom = 2
    set osd_units = METRIC
    set osd_cap_alarm = 1000
    set osd_time_alarm = 5
    set osd_current_pos = 2145
    set osd_mah_drawn_pos = 2159
    set osd_craft_name_pos = 2444
    set osd_power_pos = 2095
    profile 0
    
    rateprofile 0
    rateprofile 0
    
    
    # get ibat_scale
    ibat_scale = 220
    Allowed range: -16000 - 16000
    
    # diff showdefaults
    ,,,,,,,,,,,,,,,,
    #set ibat_scale = 400
    set ibat_scale = 220
    ,,,,,,,,,,,,,,,,,

TheAngularity, 2017-03-22:
@mikeller 
any news about `resources` ?

> yes, there still is a problem with the resource config for serial ports, probably due to the ioTags not properly being initialised. As mentioned in #2664, this should be addressed in a separate PR.

the issue is still there :(

mikeller, 2017-03-22:
What issue?

TheAngularity, 2017-03-22:
see my screenshot above /\

DieHertz, 2017-03-22:
I'll look at it tonight @mikeller

mikeller, 2017-03-22:
't was the wrong PG :-(. But fixed now.

PR #2730 (2017-03-22): Fixed display of SERIAL resources in CLI.
Fixes #2655.
closed 2017-03-22 by martinbudden