betaflight/betaflight #2656: Motor starting level problem

Warrono, 2017-03-15:
Im building my first quad, I calibrated my ESC's about a week or so ago, and I'm coming back to it to test the motors, I found that while all of my other motors start spinning at 1021, motor 1 is now not starting until around 1150. This is I assume a large issue and I don't know how to fix it. Is it a technical problem or has something happened to the ESC? It hasn't been dropped or anything so I don't know what I can do. How can I fix this?

mjedlinski, 2017-03-15:
Check all wires and solder joints from fc to esc. Calibrate again.
You can calibrate only one esc.
Maybe that esc don't enter in calibration mode at first time.

andre92, 2017-03-16:
ESC 1 lost its calibration. You need to calibrate it again, using the motors tab (disconnect battery, max throttle, connect battery, wait 2 seconds, min throttle)