betaflight/betaflight #2668: Revolt f4 in dshot600 failure to arm

BatutaBR, 2017-03-16:
I flashed a f4 revolt with betaflight_3.1.6_REVOLT firmware. With the following settings:
DSHOT600 esc protocol;
Gyro scan in 32khz
Pid loop in 16khz
Arm by switch;

I tested it connected to the pc and everything is working. If I disconnect the PC and battery the quad does not arm. If I reconnect the usb cable on the board, the quad arms. After the quad arm, it is possible to unplug the usb cable. And it stays armed and I can arm and disarm by swich. And the flight is all right. But if I replace the battery, I need to reconnect the USB cable to arm again.

I tested using multshot protocol and this does not happen.

mikeller, 2017-03-16:
Just taking a stab in the dark here: Could it be that your Vbat sensing connection is broken, and the FC detects '0V, no battery attached' when USB is not connected?

BatutaBR, 2017-03-17:
Initially I had thought of a problem with VBAT, so I disabled the VBAT function to test, and the problem continued. And also remember that after I connect the usb cable, and the quad arm, I can disconnect the cable again and everything works normal. The drone flies normal. Only if I disconnect the card completely, removing the battery and usb cable does the problem return.

closed 2018-04-21 by DieHertz