betaflight/betaflight #2670: "motor pwm frequency" parameter on dshot setup (piko blx)

orionpt, 2017-03-16:
When I select dshot the "motor pwm frequency" parameter goes away but comes back after save & reboot. I've tried re-flashing (bf 3.1.16) but behaviour remained, so I tried something that worked. Changed to multishot again and selected motor pwm synced with pid loop, saved and reboot, selected dshot and the motor pwm freq. parameter is no longer there. I flew the quad before noticing this and it didn't seem to affect performance, although I only did a short LOS test flight. Sounds like it may be a small bug so I'm adding the dump file.

[image]
[dump.txt](https://github.com/betaflight/betaflight/files/849257/dump.txt)

DieHertz, 2017-03-17:
@mikeller was it only related to Configurator?

mikeller, 2017-03-17:
There was an inconsistency in that `div.unsyncedpwmfreq` got hidden, but never shown again. Also, unsynched works for all (non PWM, despite the name) protocols, so it should always show.

orionpt, 2017-03-17:
> Also, unsynched works for all (non PWM, despite the name) protocols, so it should always show.

Was it supposed to show in dshot setup?
Does that mean that in dshot you can run unsynced ?

mikeller, 2017-03-17:
Yes. And yes.

orionpt, 2017-03-17:
So, in multishot according to Joshua Bardwell if you run motors unsynced at higher rate you would get more information that would result in an averageg command, which would be useful to reduce noise (maybe I didn't explained it the same way, but I guess you know what I mean).
Will this apply in dshot? For instance, if you run 8/8/32 will the esc get 4 redundant packets  per cycle?

martinbudden, 2017-03-17:
@orionpt , since DShot is a digital protocol with error checking, the "noise" (as you say) is much lower, so there is probably no benefit in running the motors at a higher rate.

orionpt, 2017-03-17:
ok, thanks for the answer.

closed 2017-03-19 by mikeller