betaflight/betaflight #2672: BF 3.1.6 SPRacingF3 DSHOT only 5 motor channel working

vicdzen, 2017-03-17:
Racerstar 30A ESCs.
SPRacingF3 configured for hexa. Motor connected straight to corresponding channels.
Only 5 first channels supports DSHOT (150 and 300 tested)
At the same time all 6 channels works fine in One-Multishot and PWM.
All ESC are in perfect shape (tested by swapping physically)
Remapping of 6-th channel to 7 and 8-th channel did not work in DSHOT mode (both mmix and resource remapping tested), only PWM and One-Multishot works.
Resource remapping to RSSI pin didn't work (only PWM and One-Multishot support works)
It there any kind of special steps to activate 6 channels in DSHOT mode?
Any DMA resource issue?

DieHertz, 2017-03-17:
At this point it's impossible to run 6 motor DSHOT on SPRF3.
I'll check tonight whether it's possible to rearrange like X_RACERSPI to get 6 motors, but not sure it's gonna work.

vicdzen, 2017-03-17:
Thanks for quick response!
Is it possible to sacrifice some functionality like LEDs, softserial, etc. but have 6 DSHOT enabled channels?

DieHertz, 2017-03-17:
With current `target.c` it's not possible at all if I understand correctly, but there may be a missed combination of channels which allows to do that, have to consult with @hectechfpv, X_RACERSPI target and STM32F303VC datasheet.

mikeller, 2017-03-19:
@vicdzen: Using the LED_STRIP pin for motor 6 should work, if you remap with `resoure motor 6 a08` (and obviously disable LED_STRIP):

```
DMA:
DMA1 Channel 1: MOTOR 3
DMA1 Channel 2: MOTOR 6
DMA1 Channel 3: MOTOR 1
DMA1 Channel 4: MOTOR 4
DMA1 Channel 5: MOTOR 5
DMA1 Channel 6: FREE
DMA1 Channel 7: MOTOR 2
```

vicdzen, 2017-03-20:
@mikeller I confirm that solution with 6-th motor remapped and soldered to LED_Strip pin warks! All 6 motors now can be run under any kind of ESC protocol (up to DSHOT600 tested)

closed 2017-03-20 by vicdzen