betaflight/betaflight #2673: RF one -> BF 3.1.7 - Gyro 32kHz not working

TheAngularity, 2017-03-17:
tested the last few days rf one on my revolt f4 ... after the test i'm back to bf ... but now i have problems that the gyro in 32khz mode is no longer initialized ... `moron_threshold` raise did not help ... only `moron_threshold` lower to `0` and the gyro initializes again ... is that normal?

jflyper, 2017-03-18:
Does going back (again) to RF enables 32K?

TheAngularity, 2017-03-18:
@jflyper 
if i go back to rf one, gyro 32kHz is no problem

TheAngularity, 2017-03-18:
it is already curious ... i have still tested something and found that it is not initialized if the battery is plugged in ... if it's fed only by usb everything is ok ... wtf :(( what is that now???

TheAngularity, 2017-03-18:
omg!!! ... i have found what was wrong :)) ... the magic word `current interferences` ;-)

i have also tested new o rings ... but this rings are thinner as my standard ... and the fc comes too close to the pdb ... have the fc unscrewed and see now it initialized again ... i can't believe it but so simple ;-)

ok, closed now

closed 2017-03-18 by TheAngularity