betaflight/betaflight #2674: Arming issue

jonnyvolts, 2017-03-17:
Hi all,

I'm not a new to the hobby or quads but I can't seem to get my board arming after switching from cleanflight to betaflight.

My gear is SP F3 board, x6b, evolution and eco HUBPDB. running IBUS via UART3 on IO2.

So I am trying to use aux 2 as my arm switch on my evolution but to no avail, regardless of which switch I use nothing will arm the quad. Using switches on other modes seems to work without any issue. My RX tab shows everything working as expected. The evolution TX is able to achieve 1000-2000 throttle without adjustment. I've played around with setting throttle min etc to lower and higher than the config value but still nothing.

Should my esc BLheli settings be left at default? min throttle is at 1150 and max at 1850?

I have acc off, but have tried switching it on and calibrating and then trying.

VBAT is not connected as I'm using voltage tele direct on the x6b as inaccurate as it is :)

I'm at a loss as to what I should try next.

DUMP below

# Betaflight / SPRACINGF3 3.1.6 Feb 21 2017 / 14:44:08 (1fd502c)
name Maverick
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
[... 402 lines cut ...]
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity = 55
set level_limit = 55
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650

unitware, 2017-03-17:
I can recommend the Cli command "diff" makes it a bit easy to find the non standard settings. I don't have any other good ideas though.

jonnyvolts, 2017-03-17:
# diff
# Betaflight / SPRACINGF3 3.1.6 Feb 21 2017 / 14:44:08 (1fd502c)
name Maverick

feature -RX_PPM
feature -VBAT
feature -TELEMETRY
feature -CURRENT_METER
feature -BLACKBOX
feature -TRANSPONDER
feature RX_SERIAL
feature AIRMODE
serial 2 64 115200 57600 0 115200
aux 0 0 1 1500 2100
set motor_pwm_protocol = ONESHOT42
set serialrx_provider = IBUS
set align_board_yaw = 90
set gyro_sync_denom = 1
set deadband = 1
set failsafe_delay = 20
set acc_hardware = NONE
set pid_process_denom = 4
profile 2

rateprofile 0
rateprofile 0

#

jonnyvolts, 2017-03-17:
I've refreshed a couple times and tried older firmware but to no avail. I've also tried another rx with the no change.

mjedlinski, 2017-03-17:
aux 0 0 1 1800 2000

jonnyvolts, 2017-03-17:
could you elaborate on that slightly for me :)

jonnyvolts, 2017-03-17:
Where aux 0 0 1 1800 2000 I think the change was made to aux 0 0 1 1500 2100, however with both values the aux switch is in range while active as the on state is at 2000. Still no arm though.

jflyper, 2017-03-18:
@jonnyvolts I've loaded one of my test (bare) FC with your diff, and it arms without problem.

Does your status LED tells you it has completed the gyro calibration?

jonnyvolts, 2017-03-18:
Haven't checked but enough time has passed that surely it would be complete. What am I looking for on the leds?

jflyper, 2017-03-19:
Three beeps/flashes...

jonnyvolts, 2017-03-19:
Not quite sure how to explain this but I managed to get it working through a series of reflagging to old firmware and initiating the arm switch straight after flashing after allocating the Rx type. Then making all my adjustments once the arming worked. Once that happened I then reflagged to the newer firmware and followed the same order of events and voila it worked.

closed 2017-03-20 by mikeller