betaflight/betaflight #2675: SPF3 BB log "save to file" problem

fs-fpv, 2017-03-18:
I can't save my BB logs from SPF3 fc (onboard data flash) to pc with newest BF config and Bf firmware 3.1.6. Older versions like 3.0 work, so it must be an issue from 3.1.6

mikeller, 2017-03-19:
Can you describe how it fails for you?

fs-fpv, 2017-03-21:
>Can you describe how it fails for you?

There is absolutely no reaction. If I logged a flight, I can see, that there are data on the flash. When I click "Save flash to file" nothing happens! No window, where I want to save it or any save "bar" where I can follow the progress. Nothing. Everything else works fine.

And I have another problem, on my pitch and roll axis, I have a little bounce. But only when rolling right or flipping back. I have a little video, to show the problem: 
https://www.youtube.com/watch?v=Wi4yPNQvQIM
I already played a lot with my D gains and the D setpoint weight/transition, but this strange movement is still there. I am desperated and there are only the notch filters, I can change. But without my Logs I wont make anything worse :/ So if you have any idea how to solve this, let me know.

mikeller, 2017-03-22:
There was a bug in the configurator causing this, but it's been fixed in the latest release. Can you try uninstalling and reinstalling the configurator?

fs-fpv, 2017-03-22:
I have the newest version of the configurator!. But there must be a bug with 3.1.6 too, because if I use my older quad with 3.0.1, I can download the log, but not with my new one with 3.1.6. They both are SPF3's. So that doesn't solve the problem :/

mikeller, 2017-03-22:
Ha. The fact that you get a 'Save file' dialog for one quad, but not for the other gives us a hint to where the problem might be. Did you set names on your quads? If so, what are they? (Reason for asking: Chrome on Windows has a bug in that, if a suggested name for a file isn't supported by Windows, it won't fail, but won't open the 'Save file' dialog either. And the name of the craft is used to generate the suggested file name.)

fs-fpv, 2017-03-22:
The quad where it doesn't work: 5" Alien
(I now renamed it to just "Alien" and now it works! Thank you! 👍 
The other quad is named "ZMR".
Now both work :)

To my other problem with the bounce:
Is it possible, that I have that movement on only one side because of the gyro position? It is not 100% centred. Could that be the reason and if, how can I solve that?

mikeller, 2017-03-22:
Yeah, I suspect Windows did not like the `"` in the name.

Not being centered should not be a problem - most boards do not have the actual gyro chip centered. Could it be that your board has a bit of play on one side?

fs-fpv, 2017-03-22:
No, the fc is pretty solid hard-mounted! It's the standard fc mount on the ImpulseRC 5" Alien with the carbon plate on top. So there is no movement or anything like that. It must be a firmware problem. With the older BF versions like 2.9 or lower I never had theses issues. They just came since BF 3.0.0 and I have no idea how to solve.

ronerone, 2017-04-15:
thanks for the " character not liked in the name. same with ":" here.

closed 2018-04-21 by DieHertz