betaflight/betaflight #2677: bf 3.1.6 has a wrong ibat_scale over 3.0

xlla, 2017-03-18:
FC, brf3
battery, 4s
quad, 4 motors

after flash firmware 3.1.6 and restore settings, I found current_meter_scale is missing, so I add current scale in GUI direct, then I found it was renamed to ibat_scale.

I take a test flight, export log, found a very high measured current

[image]

with same value in version 3.0.0 , it's correct.
[image]

mikeller, 2017-03-19:
I think this is a problem that is not int Betaflight itself, but in the Blackbox log viewer showing the current as 10 times what it actually was. Please reopen this issue in https://github.com/betaflight/blackbox-log-viewer/issues

xlla, 2017-03-19:
I am use blackbox explorer v2.5.10.
If I open the old log file generated by Betaflight v3.0.0, it just show correct current value, see second picture I posted.
So, I think Blackbox log viewer is fine, Or there is something wrong in BFL log format, In v3.0.0, exported log was TXT format.

mikeller, 2017-03-19:
@xlla: Yes, it is someting that was changed in the firmware between 3.0 and 3.1, and the log viewer does not correctly take that into account yet. => https://github.com/betaflight/blackbox-log-viewer/issues

xlla, 2017-03-20:
@mikeller Alright, I see issue #47, #48 on project blackbox-log-viewer are around this problem, I will open this issue in on that project.
thank you!

closed 2017-03-20 by xlla