betaflight/betaflight #2680: Sparky 2.0 Yaw Issue

Tomcatt, 2017-03-18:
Did a test flight in the wind and I like your default PIDs.  This old FC was looking really good.

Okay, everything is working great but my yaw is being read.  I can arm it (but I use a switch) so that's not a problem.  On the receiver tab, I can see it working just fine but it's a no-go after arming.  On the ground and in the air, I have no yaw.

Is there anything I can do on my end first?

MJ666, 2017-03-18:
Did you load an old cli dump? Please try to configure from scratch.

Tomcatt, 2017-03-18:
No, but I can wipe it and start again.

MJ666, 2017-03-18:
Usualy this effects you can get if you  add old config. If this was not done the new config likely will not help. What BF release are you using? What kind of RX coniguratio you are using?

Tomcatt, 2017-03-18:
Just finished flying and still no change.  (Still fun minus one axis) 

Thursday's 3.2.0 build.  This was the same problem I had a week ago with no inputs.  Now it's just the yaw.

RX?  it's the S.Bus XSR TAER1234 config

DieHertz, 2017-03-18:
`rc_controls.c` is missing an initializer for `yaw_control_direction`, hence it's 0.
Also, it's possible to set it to 0 from CLI, maybe should use a lookup table.

Tomcatt, 2017-03-18:
What number should it be?

DieHertz, 2017-03-18:
It should be 1 or -1 for reversed props, CLI doesn't allow any other value

Tomcatt, 2017-03-19:
Okay, will try it out as soon as my transmitter battery recharges.

closed 2017-03-19 by Tomcatt