betaflight/betaflight #2681: Motor PWM problem

Serj1032, 2017-03-18:
Hello!

I have Eachine Racer 250 witn [flight controller ](http://img.banggood.com/thumb/water/oaupload/banggood/images/3D/42/e9f8aae7-97e6-2292-bf88-1508ea5d5bea.jpg)based on STM32F103CBT6.

I flashed betaflight 3.1.6. [This is](https://cloud.githubusercontent.com/assets/6850581/24074221/d10cb0c8-0c1d-11e7-9a15-bcd60e138067.png) my configuration. Motors map on PA8, PB7, PB8, PB9

1. This PWM signal at motors 1,2,3
[image], but the same time at the motor 4 PWM is
[image]
Why?

2. When I start (throttle at middle) and stop (throttle at min) several times, some motors switched off. And when I put throttle at middle next time, this motors are not rotate, but other start rotate. What is the problem?

Thanks!

borisbstyle, 2017-03-18:
As far as I know there is no offical target for that board so its hard to review the code for it

Serj1032, 2017-03-18:
@borisbstyle , my flight controller very similar at the CC3D. I'm about PWM input from receiver and output to ESC. But I'm agree with you, that this device is no official, and if you can't advise me smth, I'll be forced to throw it away))

closed 2018-04-21 by DieHertz