betaflight/betaflight #2683: AIOF3_BRUSHED Flight Control Board PPM/AFHDS 2A/1A compatibility

osgood232, 2017-03-18:
I've recently bought an Eachine AIOF3_BRUSHED Flight Control Board Built-in OSD Frsky/Flysky/DSM2/DSMX Receiver (running Betaflight 3.0.1) board anticipating it to work with my ppm setup. However I have noticed that it is AFHDS 2A mode. I was anticipating some for of backward compatibility the v1, but alas there seems not to be.

I was wondering if anyone knew the exact difference between the two versions (AFHDS V1 and V2)and why the board doesn't support AFHDS V1? Also if anyone is anticipating a future change to support this.

One might ask why I'm running an old protocol, well the answer is simple - All of my rc kit is set up as ppm and off one single controller. Which includes quite a few hardware modifications/hacka. I'm also happy without features such as telemetry as additionally I only fly FPV and get most of the information fed back to me via my goggles.

Although I'm new the the forum I have experience in both electronics and software so feel free to get as techy as you like with any answers.

Also apologies if this has already been discussed, but I could not find a relevant thread about it.

Thanks

Paul

osgood232, 2017-03-25:
Hi marcroe,

Thank you for your response, yes it does seem like it should be at the receiver/transmitter end however both myself and a colleague (we are both engineers) have have been looking at this in detail. the board use a transceiver chip called the A7105. This is quite configurable over SPI and can be used as either a transmitter or receiver) however we cannot seem to find any options on the datasheet that references this mode. I have already scoured rc groups for the same issue and as there was a distinct lack of information we only concluded that it was handled at the software level.

You are the first person who I have had direct  contact over this issue so if you are confident that it isnt configured at software level then I can only conclude (like you) that it is managed at chip level - I just need to keep looking at the datasheets :)

Thanks for your time

Paul

martinbudden, 2017-03-25:
@osgood232 , I'm not familiar with the flight controller you speak of. From your post it seems you chose the Flysky option. For an AIO board the receiver will be connected to the flight controller via serial, so you will need to set the receiver to serial and to use IBUS (the serial protocol used by Flysky). You will also have to set up the correct UART - your board documentation should tell you which UART the receiver is connected to.

osgood232, 2017-03-25:
hi, Its an all in one board with integrated receiver. its for micro FPV quad which is my area of interest so there is not many options. Its already purchased as the flysky option but only the 2A PPM format (which is frustrating) I am looking at the best and most efficient way to make it ppm v1 compatible, it might be a software change or it might be a hardware change, either way im hoping that I can find a way round this without changing my controller setup.

paul

mikeller, 2017-03-26:
Hi Paul. As @martinbudden pointed out, Betaflight does not currently have support for driving RF chips over SPI for the FlySky protocol. So if the board in question is running Betaflight (and it is not a version that has had unofficial and unpublished code added by the board's manufacturer), then there must be another MCU on the board that drives the RF chip over SPI and outputs the serial signal that is then read by Betaflight. For other protocols like FrSky or openLRSng, this RX MCU is often a ATMega 328. The code that is run by the RX is often proprietary, so there might be no way to modify it short of reverse engineering.

closed 2018-04-21 by DieHertz