betaflight/betaflight #2691: Betaflight FC will not communicate?

HussoFPV, 2017-03-19:
Hi all, I Recently built a new Quad with the one and only BetaflightFC (Finally!).

Everything is complete with no shorts (I always check continuity before first power up).

All working well until I tried to flash the latest BF on the board. I have two other quads, running a DODO and DTFc Flight Controller, so I am familiar with the drivers required and the use of Zadig for DFU flashing.

However My BFFC will not communicate with my Laptop, I keep getting the "USB Device not Recognised" when I plug it in. On BF it states "Failed to open serial port".

Its been a solid 6 hours of research, stress and failure. I have also tried this on other Laptops, two are Windows 10 and one Windows 7. The windows 7 Laptop manages to communicate with the board only in DFU, So I was able to flash BF 3.1.6, however, even with the latest CP210x & STM USB VCP Drivers correctly installed on all Laptops, I am unable to get any comms in the BF Configurator, even after the latest Flash (Target used: BETAFLIGHTF3 and current Configurator: 3.1.1).

I have uninstalled ALL previous drivers and re-installed, still nothing. 

My Device Manager shows "Device Descriptor Request Failed" under USBC.

I've ran out of Ideas and starting to think that my board is defective, but I want to eliminate every possible cause before I say that it's the FC at fault!

Has anyone else had a similar issue with the BFFC?

Any help or information would be greatly appreciated. Thank you in advance.

Kind Regards,
HussoFPV

jflyper, 2017-03-19:
Did it connect with the configurator before flashing?

AndersHoglund, 2017-03-19:
How much extra stuff have you connected to the FC and powered from the +5V rail? 

There is a design flaw on BFFC (at least on my early sample), there are no isolation diode on the main 5V rail. This means USB is powering "the whole world". I have for instance a LED-strip and a beeper on my BFF3. Both those turn on when only connected to USB. This causes intermittent drop-outs on my laptop because this draws too much power. I have to power the FC from a battery to have a stable USB connection. Please try that.

This has been reported on RCG some time ago. https://www.rcgroups.com/forums/showpost.php?p=36655239&postcount=816

HussoFPV, 2017-03-19:
No it didn't connect before the first reflash.

The only thing I have powered through the 5v rail is my receiver. I have tried connecting with the battery plugged in but this makes no difference.

mikeller, 2017-03-19:
I think at this point it is fair to make the assumption that this is a hardware fault, and to invoke the warranty.

bluebluby, 2017-03-19:
Try to power your quad with a 2s or a 5v power supply and then to connect to usb.

HussoFPV, 2017-03-19:
Still nothing. I believe it is the board that is defective. I will contact the supplier and get a replacement under warranty

HussoFPV, 2017-04-04:
Confirmed the board was at fault.

closed 2017-04-04 by AndersHoglund