betaflight/betaflight #2693: Beeper on aux channel tied to ON_USB not RX_SET

ethan4, 2017-03-19:
I would like to have the beeper enabled by switching a channel on my RX but not wake up my housemates every time I plug my quad into the computer for some tweaking and based on the notes here (https://github.com/betaflight/betaflight/blob/master/src/main/io/beeper.h) it looks like this used to be triggered with RX_SET rather than ON_USB.

mikeller, 2017-03-19:
ON_USB controls the beeper in the case when the board is connected to USB. `beeper -ON_USB` will disable beeping in this case, except for init beeps that happen before the config is loaded.

ethan4, 2017-03-19:
@mikeller didn't test this before but seems like only stops beeping on aux channel, while you are connected to USB, so my testing wasn't working but connecting lipo power and testing again shows it works correctly.

closed 2017-03-19 by ethan4