betaflight/betaflight #2700: eachine racer flight issue

fabold, 2017-03-19:
i have an eachine racer 250 with cc3d and betaflight today while i was flying it starterd tilting right and i lost control. i changed the prop that i broke and i was back on flight but it happened again and the second time i broke the battery plug. please tell me how to fix that.

mikeller, 2017-03-19:
Check prop nuts, motors, ESCs, cabling / solder joints on your right hand side.

Cheredanine, 2017-03-20:
And make sure quad is flat and stationary when you plug the lipo in

lostcontrol, 2017-03-20:
Also drifting in level mode is often caused by vibrations. If the drift gets worst with time, this is definitely your issue. Again, check motors and props balancing and/or soft-mount your FC.

closed 2018-04-21 by DieHertz