betaflight/betaflight #2703: Flight Mode lockup

MontanaJared, 2017-03-20:
I don't know how often this would actually happen, but I have a second switch that overrides any flight modes for failsafe (Taranis).

I noticed that if I switch the switches (primary flight mode switch and failsafe switch), I can get Betaflight confused and it locks to one of the modes. Further input is ignored and no more led's flickering when mode is changed.

Maybe nothing, but has anyone else tried this? I am not sure where the threshold is, like how fast you flip the switches or if its number of times.... but getting stuck in a mode is disconcerting.....

J

mikeller, 2017-03-20:
Is the switch that you are activating the failsafe switch?

If so, this is exactly how the failsafe switch is supposed to work: It simulates a failsafe event (TX connection lost), and it is designed to stop the craft from accepting any more input before it is power cycled. Read https://github.com/cleanflight/cleanflight/blob/master/docs/Failsafe.md .

MontanaJared, 2017-03-20:
So, by flipping the switches like a madman, I triggered the FC failsafe? Which, yes, you are correct, I had to power cycle everything to connect again. Is it only for convenience then that when we flip the failsafe switch to test it while connected to BF that it doesn't lock up like this? So, in effect, the crazy flipping switches caused the FC to say "Yes, this is the real thing".... and lock down?

Thanks for the reply.... and the education....

mikeller, 2017-03-20:
No, the failsafe switch will always simulate a failsafe event. The difference is probably that when the craft is not armed, the failsafe event does not require a power cycle to recover.

SwiftingSpeed, 2017-03-20:
I think we need to know exactly what your "failsafe" switch on your radio do, it isn't clear what you mean by override.
1. Does it activate one of the "normal" flight mode? like angle mode for bailout during flip practice
    For example one mode switch for changing between acro/angle/horizon and the failsafe switch to force the mode into angle regardless of the mode switch position.
2. Does it activate the failsafe flight mode under the modes page in configurator?
    The failsafe mode on the modes page in configurator has a range configured that your switch is activating.

pupojmastro, 2017-04-03:
I have a problem that sounds a lot like these. Leading *theory* is, yes, FC failsafe gets triggered and quad freaks out. First step is to set failsafe in Betaflight. But that only stops the freak out. Strongly suspect a receiver issue. Signal is getting dropped. That, I still don't have an answer for.

closed 2018-04-21 by DieHertz