betaflight/betaflight #2704: 3.1.6 and SPRacingF3EVO ports tab bug

MylesShannon, 2017-03-20:
I think the image below is pretty self-explanatory. Firmware version 3.1.5 does not have this issue. I'm running configurator 3.1.1 and I can't select certain drop downs for the first UART1 because they revert back to "Disabled" after "Save on Reboot"

[image]

Hope this helps!

mikeller, 2017-03-20:
No it isn't self-explanatory. The way the firmware works is that, when saving the configuration on the flight controller, if it detects a combination of settings that won't work, it will reset these settings to default. See https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md#constraints

MylesShannon, 2017-03-20:
The port settings resetting to default is only a side effect of the bug I was trying to outline. Did you take a look at the included picture? There's three UART1s listed. This is obviously not as intended and a bug I've only found in build 3.1.6

borisbstyle, 2017-03-20:
Spracingf3 evo doesnt have softserial enabled so the issue may be coming from there

MylesShannon, 2017-03-20:
The SPRacingF3EVO got softserial. Not sure under which release, but I've used it previously on this board.

https://github.com/betaflight/betaflight/blob/master/src/main/target/SPRACINGF3EVO/target.h#L72

mikeller, 2017-03-20:
Ah sorry no, I didn't study the picture long enough to spot it. Maybe next time use red circles or suchlike to mark what you want to point out.

The two extra UART1 instances at the bottom are in fact the result of SOFTSERIAL being disabled in 3.1.6 for this target (due to size constraints), but being left included when specifying the total number of ports. Just ignore them.

carloscampos775, 2017-03-25:
I have the same board with the same problem if you set every UART1 to (example: smartport) hit save it will work.

carloscampos775, 2017-03-25:
Also, I opened this bug in #2592

mikeller, 2017-03-26:
Duplicate of #2592.

closed 2017-03-26 by mikeller