betaflight/betaflight #2708: travis build is green even when failed

unitware, 2017-03-20:
see: https://travis-ci.org/unitware/cleanflight/builds/210546524

using a custom travis.yml that only specifies a test goal and a few targets
```
env:
#  - PUBLISHMETA=True
#  - PUBLISHDOCS=True
# Specify the main Mafile supported goals.
  - GOAL=test
#  - GOAL=all
# Or specify targets to run.
#  - TARGET=AFROMINI
#  - TARGET=AIORACERF3
...
#  - TARGET=MOTOLAB
  - TARGET=NAZE
#  - TARGET=OMNIBUS
...
#  - TARGET=SPARKY2
  - TARGET=SPRACINGF3
  - TARGET=SPRACINGF3NEO
#  - TARGET=SPRACINGF3EVO
...
```

```
%% stm32f10x_spi.c 
%% stm32f10x_tim.c 
%% stm32f10x_usart.c 
%% stm32f10x_wwdg.c 
./src/main/telemetry/telemetry.c: In function 'telemetryCheckRxPortShared':
./src/main/telemetry/telemetry.c:115:12: warning: implicit declaration of function 'rxConfig' [-Wimplicit-function-declaration]
         && rxConfig()->serialrx_provider == SERIALRX_IBUS) {
            ^~~~~~~~
./src/main/telemetry/telemetry.c:115:22: error: invalid type argument of '->' (have 'int')
         && rxConfig()->serialrx_provider == SERIALRX_IBUS) {
                      ^~
make[2]: *** [obj/main/NAZE/telemetry/telemetry.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory `/home/travis/build/unitware/cleanflight'
make[1]: *** [binary] Error 2
make[1]: Leaving directory `/home/travis/build/unitware/cleanflight'
make: *** [NAZE] Error 2
The command "./.travis.sh" exited with 0.
cache.2
store build cache
0.01s
2.13snothing changed, not updating cache
Done. Your build exited with 0.
```

AndersHoglund, 2017-03-20:
What does your .travis.sh look like? It seems to throw away the return value from make and return zero to travis:

    "./.travis.sh" exited with 0.

The "normal" travis executions do detect failures....

unitware, 2017-03-20:
It is the same travis.sh as in v3.1.6

https://github.com/unitware/cleanflight/blob/5d2be6e03b8cca7fc019e0470086d0749910fc0c/.travis.sh

I think it works with GOAL: test and GOAL: all, but maybe not with TARGET=nnn

AndersHoglund, 2017-03-20:
Probably the mix of both GOAL and TARGET... Try using GOAL only, i,e. GOAL=test and GOAL=NAZE ... etc 
etc. The old semantic "make TARGET=NAZE" vs the new simpler "make NAZE".

unitware, 2017-03-20:
ok, so is TARGET not used anymore? shall I remove those parts in a PR?

AndersHoglund, 2017-03-20:
Or fix it.

unitware, 2017-03-20:
Yes but If it's old semantics that is not used in travis, the cleaner way would be to clean. However I do not know all uses of the publish stuff. Is this used somewhere?

unitware, 2017-03-20:
..and target: stuff

closed 2017-03-23 by AndersHoglund