betaflight/betaflight #2714: Feature request - GPS coordinates display in Betaflight OSD

dovcharov, 2017-03-21:
Good day, team.
I'm new to FPV flying, currently I'm building my first FPV quad on Omnibus F4 AIO (with integrated Betaflight OSD) flight controller.
I have little expreience in flying, I wanted to set up a GPS module on my racing quad to get GPS coordinates on my OSD. This way, if I accidently lose control of the quad, I can find it by GPS coordinates on video recorder. I successfully connected GPS module to my FC, and enabled it in Betaflight. But, to my dissapointment, there's no option to enable GPS coordinates display in Betaflight OSD.
Can you please implement GPS coordinates display in Betaflight OSD? As I understand, 3rd party MW-OSD modules have such function.

Demilliac, 2017-03-27:
I'd love to have gps display on the integrated osd. It will be a killer feature for long range runs.

mikeller, 2017-03-27:
Have you had a look at iNav?

dovcharov, 2017-03-28:
INav has such feature, you are right. But we are talking about having it in Betaflight as well. My justification is above.

mikeller, 2017-03-28:
Just making sure that you're aware of the options.

jflyper, 2017-03-29:
BF3.1 + OMNIBUS F4 AIO is not capable of RTH function because it does not support I2C atm.

jflyper, 2017-03-29:
My plan was to import the UART3/I2C sharing code from iNav, but recent change in UART initialization code made it a bit hard to do this.

Simply enabling I2C2 kills UART3, because I2C error handling code turns on the uninitialized bus upon compass probing (at least it was like this when I gave it a look last time); it is not a solution for generic configuration for this board.

dovcharov, 2017-03-29:
You have read my request incorrectly. I did not request RTH, i2C etc.

**I asked about adding an option to Betaflight OSD tab, to display GPS coordinates in OSD.** 

So that when you fly in FPV goggles, and your copter flies away, you could review the video, and see the last GPS coordinates of the quad. It's much easier to go and find it in the forest in such case. It's a noob feature - it is already present in iNav 1.6, and in MW-OSD, so importing it to betaflight should not be a hard task.

jflyper, 2017-04-03:
#2771

Zethyl, 2017-05-03:
This is a must have feature! I have completely avoided betaflight OSD because of it. Please implement!

mls228, 2017-05-03:
This is a little off topic but here is my iNav on Omnibus F3.  
https://youtu.be/GY6YZty-yAQ

red-barran, 2017-05-23:
I'd would also love to see coordinates in the OSD
Adding GPS altitude would also be great!

ABLomas, 2017-06-11:
Inav is missing some features for fun flying (fpv angle mix, dshot support, etc); betaflight is still better if you do not care about RTH.
But coordinates are usefull for finding copters (no need for RTH). +1 to this.

mikeller, 2017-06-11:
@ABLomas: For this purpose, having the coordinates reported over telemetry, and using logging on your TX, is a lot more useful than seeing them in your goggles while you are flying. And this feature is already implemented.

ABLomas, 2017-06-11:
Telemetry requires capable TX/RX, which usually means short-range frsky (usually < 2km range); and for long-range flying ppl are using LRS, most of them do not have/use telemetry.
Also, checking coords on TX vs seeing them on OSD... Guess what is much easier?
Anyway, not big issue for me - i have enough spare UART's so can use one for MW-OSD.

mikeller, 2017-06-11:
@ABLomas: So you use Betaflight for long range flying? You should seriously consider switching to iNav. Also, n.b., GPS coordinates over FrSky telemetry works just fine on openLRSng. If your long range system doesn't do that, you should think about upgrading. 

> Also, checking coords on TX vs seeing them on OSD... Guess what is much easier?

When you are crashed? Checking them on the TX - your OSD will only be displaying static at that point.

Also, I never said that this shouldn't be implemented. I just tried to show you a more reliable way to get the GPS coordinates to retrieve your craft after it's crashed.

ABLomas, 2017-06-11:
I really don't understand your point.
No, i do not want inav on copter, i simply do not like it. I use inav on 10+ plane setups and it's fine but i do not like inav on copter.
Also, OLRS range sucks with telemetry. Same goes for most LRS systems, unless you want high-power RX onboard, which will cause more harm via interference (can't do proper spacing on copter). No telemetry transmission -> better for RX, also better for electronics. Been there, done that for years and i think, i know how stuff works. Wanna beat this? https://youtu.be/kJSHNC-yuyc
Also, adding MW-OSD board instead of replacing all LRS system with tens of RX'es is much, much cheaper and simpler, just costs some weight...
And coords on RX aren't reliable, unless you are logging everything to SD card, then you still require full replay from card, etc etc - compared to OSD+DVR setup is way too complicated...
Anyway, all that is offtopic and reminds me old times when someone asks "how to change start menu color" and someone suggests "install linux" ;-)

mikeller, 2017-06-11:
Fair enough. Looks like your setup is so far out there that there is probably not many others using something similar. My remark was more geared towards the average user who would not want to spend money on a grainy DVR just to be able to get some coordinates if they can just use telemetry and logging for free.

red-barran, 2017-06-11:
From previous recommendation of Mike (in #2442) to look at INav for GPS, I looked into it. It just does not seem to be an option for me. The number of targets is limited. INav's documentation say that for GPS features on a Quad, GPS is needed obviously, and so is a barometer and a compass (magnetometer). Baro and Mag ALL seem to talk on I2C. Not many boards have I2C, or have a built in baro. The new FlyTower F4 seems to have Baro (optional) and I've been looking at one of those (but they seem to have been pulled in the last day or two!)

To use INav, basically I'd have to bin the entire quad as none of the recommended or marginal INav boards are suitable for my quad, i need an AIO like the BF F3, the DYS F4 or the F4 Flame - that include a PDB. A lot of new super compact quads like the DIATONE CRUSADER GT2 for example have the same constraint. Also, for better or worse I like the Turnigy Evolution transmitter because it is so much smaller, cheaper and more portable than the Taranis so the extensive telemetry isn't an option. Battery and RSSI telemetry seems fine. Using the Evolution also means that the receiver choice is limited, both the longer range options are quite large unfortunately (impacting available space). In 2442 I learned that there is no GPS for F3, so what I did after Mike pointed that out was went and bought an F4 board, so I do now get speed, but the altitude data in OSD is limited to Baro only (and the 4 supported options all talk I2C). This is why I'd love to see the firmware add GPS position (and altitude) to Betaflight (which is what works best for me). 

Mike has pointed out that Betaflight is focusing on racing. Fair enough, although I would think that most racing things do care about how fast they are going, and the ability to locate a lost quad from DVR OSD coords is great insurance in addition to a beeper (like ABLomas, I record the transmitted video - because I don't want a massive camera on my quad). Another factor is that most jurisdictions have limitations on height so having more flexibility to allow quad pilots to remain compliant with the law would be just fantastic.

Sorry for the long story. I like racing quads, I love Betaflight, the addition of a tiny GPS receiver for basic speed, height and position really seems to make sense

martinbudden, 2017-06-11:
@mikeller , I'm persuaded by the arguments. I think we should try and get this in 3.2

jflyper, 2017-06-11:
@DanNixon You are summoned ;)

mikeller, 2017-06-11:
@martinbudden: I never said it should not be done. I just tried to point out alternatives that are available, and work with the hardware / software that a lot of users already have. There is nothing wrong with having more options for data to be displayed in the OSD, and since we already got the data, it won't use much flash space.

@red-barran: You make a good point, iNav won't work with a lot of the boards that Betaflight works with.

DanNixon, 2017-06-11:
@jflyper What needs done? Position and satellite count are there, home direction has a PR.

I guess distance to home would be useful too.

jflyper, 2017-06-11:
@DieHertz 
Please checkout the latest firmware and tell @DanNixon what to add.

red-barran, 2017-06-12:
Really appreciated guys. How can I get involved in testing?

mikeller, 2017-06-12:
@DanNixon: I guess vertical distance (i.e. altitude above ground) would be nice.

DanNixon, 2017-06-12:
@mikeller Part of me thinks GPS altitude should be handled in `calculatedEstimatedAltitude()`, that way the existing altitude element (and altitude hold I assume).

mikeller, 2017-06-12:
Good point.

Zethyl, 2017-06-12:
The compass heading should also be available to display on the OSD in addition to coordinates. This has become useful for me on fixed wing after I lost video, that way I knew the approximate direction it was going at said coordinates.

peteoz, 2017-07-18:
for all the valid reasons given above I'm glad to see this is getting implemented. I like to take my performance quads (BF FW) to the mountains and do some long range flights, wouldn't want to run iNav either. TBS Core Pro with GPS module does the job (in terms of getting all the GPS info on OSD) but it's just not practical to fit on "normal" size quads, with TBS gear you would typically need something like 7inch Alien.
Anyway, I understand F4 FC is required, am I correct to assume the FC needs SDA & SCL pads and a free UART to run Ublox M8N ? What FC would you recommend (one that also comes with BF OSD). I was looking at Lumenier F4 AIO FC but without documentation I'm unable to tell whether there are SDA and SCL pads.

dovcharov, 2017-07-19:
You don't need SDA and SCL pads for GPS solely. SDA and SCL are needed for compass module built into GPS modules like BN-880 or UBLOX-M8N. Check Omnibus F4 AIO - it has SDA/SCL pins.

Please note. Betaflight does not support autonomous (GPS and Compass-assisted) flights. Betaflight focuses on racing performance, and GPS support in it is rudimentary. 

My initial request for adding GPS coordinates display to OSD was related only to the requirement to look for missing copter using video recorded by FPV googles. However, currently, I don't need this feature anymore. I bought a loud buzzer, which allows me to locate the quadcopter easily. 

If you are looking for GPS-assisted or autonomous flights (GPS stabilization, Return to launch) - please consider following:
1. You definitely need a flight controller with barometer! Consider Omnibus F4 Pro, or any other FC with built-in barometer. 
2. Also some flight controllers have barometer AND compass built-in. This is NOT good. Built-in compass is useless because of high magnetic interference, and also it blocks external compass. Consider flight controller with barometer but WITHOUT built-in compass.
3. You need to deploy your GPS/compass module away from quadcopter electronics to prevent magnetic interference. Consider buying a GPS mount mast. 
4. Best GPS modules are Ublox M8N and Beitian BN-880. These are essentially one and the same GPS/compass chip. Only BN-880 does not have a case, and M8N - does have.
5. Do not consider betaflight as your autonomous copter firmware. It does not support autonomous flying. Consider INav or Cleanflight.
6. For better autonomous flying experience, consider APM-based line of modern flight controllers. Pixhawk PX4 is a little bit outdated - new F4-based Pixhawk is called Pixracer. It supports latest Pixhawk firmware, and it's autonomous flying features are top-notch. But it's not good with racing/freestyle.

peteoz, 2017-07-19:
@dovcharov Thanks for the detailed info. I only build freestyle/acro quads and I'm not interested in autonomous flights. iNav is not suitable for my flying needs, I run only Betaflight FW. I can fly even with my long range 5inch quad 2.5km away (and my 6inch or 7inch get even further) and losing the quad at that distance in difficult terrain would make it very difficult to find without last known GPS coordinates. Yes, one can retrieve the GPS info from the Taranis screen (last known coordinates will stay on) but if you accidentally switch off the radio then you need laptop with you to retrieve the info from the logs whereas it's much simpler to replay the footage from the goggles DVR.
Altitude above ground would be also quite useful to have while flying, as well as home direction - sometimes at such distance away it's easy to get disorientated and having an arrow to show home direction would be great thing to have.
I'm now looking at buying Matek F405 AIO F4 FC, it has 5 UARTs, BF OSD and SDA/SCL, I only wish the gyro was MPU6000.

dovcharov, 2017-07-19:
Exactly that's why I requested this feature. I understand your case.

Note, that for altitude display in Betaflight, you need barometer as well. I connected GPS module to betaflight FC (my FC is witout baro), but I never saw any altitude (at least in Betaflight up till 3.1.7). It does not know yet how to take GPS altitude, and display it in OSD. maybe version 3.2.0 will have some new code for GPS-based altitude - don't know for sure

New Matek FCs are very good. There's new Matek FC coming which will integrate PDB and F4 FC in one board! For super-slim builds. And guess what? matek will soon release TBS Unify clone - a VTX with Pitmode and SmartAudio support. Hurray to Matek!

Other options with MPU6000 are DYS AIO FC, iFlight Poseidon/Myairbot Asgard AIO FC (these ones even have 4-in-1 ESC integrated).

There's also a new flight controller called Holybro Kakute F4. It's based on new ICM20689 32K  gyro, which is less noisy than it's other 32K gyro brothers. Also, the gyro is already soft-mounted on Kakute F4 FC, so you don't need to soft-mount flight controller itself. You can reach 16K/16K PID loops on this FC easily.  But it does not have baro.
 
As for myself, for new quadcopter build I ordered Omnibus F7 flight controller. it has both - MPU6000 gyro, ICM202602 gyro (needs softmount) and barometer + 3 UARTs + hardware inversion.

DanNixon, 2017-08-21:
Everything that was requested here has been done.
Can this be closed now?

red-barran, 2017-09-04:
Hey there Devs, LOVE the new 3.2.0 BF (RC4 at time of writing).
Can I ask if the GPS altitude was implemented? I can see in the OSD tab that GPS coordinates, direction to home, distance from home (and the speed/satellites is still there like it was in 3.1.7). Can't spot GPS altitude, just the Altitude that was there in 3.1.7 which is baro. 
Thanks

DanNixon, 2017-09-04:
@red-barran No. Ideally the altitude indicator should pick (or fuse) a suitable source, this probably isn't being done at the moment.

mikeller, 2017-09-04:
@DanNixon: The altitude estimator is a mess. I think for 3.3 we should consider 'borrowing' the implementation from iNav.