betaflight/betaflight #2717: FrSky XMPF3E - DSHOT Motor #4 Problem

x-sergo, 2017-03-21:
I am experiencing a problem with the new Aikon SEFM30A 4-in-1 and DSHOT on the FrSky XMPF3E flight controller.

Using OneShot/MultiShot is working fine. If I switch to DSHOT motor #4 is not working.

Can someone confirm that the FrSky XMPF3E needs a new pin assignment for motor #4 as the FC is using the SPRACINGF3EVO firmware?

Bes

mikeller, 2017-03-22:
Read https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol#spracingf3evo----must-move-motor-4-to-new-pin-assignment-cli--resource-motor-4-a06-then-solder-esc-for-motor-4-to-motor-output-5-fixes-dma-conflict-with-motor-outputs-2-and-4

dronehitech, 2017-03-22:
Read it already :). That is why I have asked if someone who has already tested it, can confirm it has the same issue as the SPRACINGF3EVO board.

mikeller, 2017-03-22:
Well if the board in question is using the SPRACINGF3EVO firmware then it has to be the same, as the pin assignments are the same (if they weren't, the firmware would not work properly).

dronehitech, 2017-03-23:
Ok, motor #4 signal wire to pin for motor #5 connected. Pin assign over CLI and now everything works as it should be with DSHOT as well.

closed 2017-03-23 by dronehitech