betaflight/betaflight #2722: acro mode increase energy consumption/drains battery enormously - airmode does not! why?

matahata, 2017-03-22:
Hello!
I’m using SP3 EVO BRUSHED.
I've made the experience that when I fly in Airmode, my craft fly fast, raise fast in height and the vbat doesn’t cause the beeper to sound, that's fine.
BUT switching in rate/acro mode, the quad is slow and beeper starts beeping right after start, with full loaded batteries. 
What setting could be wrong?
Thanks for any ideas!

Yamaford, 2017-03-22:
Airmode is not a flight mode. Rate, Horizon and level are flight modes.  Do you have one of these modes assigned? If not you are already in Rate by default.  RCgroups thread would be a better place for these type questions as I do not think they deal with the operation of the software but more to your specific setup.

matahata, 2017-03-22:
I'm sorry, it was "Horizon" mode! Thanks, I will go to a RC group, if I'm totally wrong here.
Hmm, I thought, to "calculate" the Horizion mode (inclusive "Accelerometer") it will need more power, so the draft have less power for rising, turns, and so on.. but Horizon mode causes my quad to fly faster than Rate mode (which I expected to be at least a tiny bit the opposite. In rate mode only the gyro are necessary, if I'm not wrong.) As soon as I switch into rate mode, the quad drops on speed, it feels like clumsy in comparison to Horizon mode. I flew a lot of rate mode before, so that is not the point.. Could it be, that the PID settings are better in (automatic) Horizon mode, than the one, that are set for the Rate mode? What else can change/switch on/off??

matahata, 2017-04-27:
If anyone is interested. Here my assumptions:
1) Too small props for my quad, therefore only in "Horizon mode" causes the quad to stay "horizontally balanced" enough in the air and to gain height quicker. In acromode I was never "horizontally balanced", therefore I lost height easily. The quad with the too small props was really fragile regarding this point. --> Larger props, better results.
2) Too weak batteries for the build. I had to buy new batteries with higher capacity to have fun this build
3) The battery meter of the mainboard is crap, I could compare the values of a fully loaded batteries versus the one shown "live" within betaflight -> big difference. I could use software calculation compensation, but I finally switched it completely off

closed 2017-04-27 by matahata