betaflight/betaflight #2726: Flytower F4 v2 led not working only buzzer

asaerong, 2017-03-22:
Hi, connecting via Led+Buzzer connector and the led is not working, just no lights, only buzzer works. Tested with 2 units, the result is the same. 

If connected via USB only, sometimes one led dimly light red. Could not get D-in directly via pwm5 because the board is built for ppm or sbus only.

Hope it can be fixed. thanks in advance.

mikeller, 2017-03-22:
I assume that what you are using is one of those combined LED and buzzer boards that can use a variety of input sources?
If so, these need to be configured properly to accept input in the WS2811 format that Betaflight uses for the LED_STRIP output. Please refer to the manual that came with the board.

closed 2017-03-22 by mikeller

asaerong, 2017-03-23:
Hi, thanks for the reply, Yep, its Matek combo led+buzzer board which already configured to accept the FC input in WS2811 format that Betaflight uses. 

Forgot to mention that the same led+buzzer board has been used in 2 of other build using Omnibus F3 with no problem, and another 2 with Omnibus F4 without SD, but in F4 it has to use rx pwm 5 instead led port. 

It might be the same case with this board... unfortunately it has no dedicated pwm5 input to take the led feed from.

mikeller, 2017-03-23:
Ok. OMNIBUSF4SD defaults to using PWM 6 as the LED output. Have you tried that?

asaerong, 2017-03-23:
I would love to try, unfortunately in this board there is no physical input for pwm 6 or 5. Only sbus/ppm connector.

mikeller, 2017-03-24:
So from what you are saying, on the board that you have there are only pins / pads for RX and 4 motors. Obviously this means that there is no place to connect an LED strip to, and there is nothing that can be done in software to change this. Or are there any other pins / pads on the board that are connected to the MCU?

asaerong, 2017-03-24:
Here is the board: [link to Betaflight wiki](https://github.com/betaflight/betaflight/wiki/Flytower-F4-v2), it's an AIO (including vtx) board with OMNIBUSF4SD as target. It has pin based connection to stackable 4in1 esc (and from there to motors). a cam port, rx input port (which is only sbus/ppm, 5v and ground) and ... led+buzzer port which is 5v, ground, D-in and B+. It has no other pins/pads for pwm input. Only accept sbus/ppm.

[image]

mikeller, 2017-03-24:
So it looks like it **does** have an LED output, on the connector next to the SDCard slot. The instructions on the wiki say that you should be using the FISHDRONEF4 firmware. Have you tried if the LED strip works with that firmware?

digaus, 2017-03-24:
The Wiki Link is wrong ... that is the v2 he is using V1

digaus, 2017-03-24:
What I suggest is taking a multimeter and check  to which pin on the F4 chip the LED pin goes and then just remap it

mikeller, 2017-03-24:
Ah. In that case, maybe @fncisco can clarify what MCU pin is connected to the LED pin on V1. Or use a multimeter.

asaerong, 2017-03-24:
Ok thanks guys for looking into this. will try the multimeter check on the F4 chip, in the meantime, hopefully @fncisco can chime in on this. Otherwise we can conclude that this board (V1) does not support led for now.

fncisco, 2017-03-25:
@asaerong the led strip is conected to PD2, is pin 54 of stm32f405rg. let me check where is on omnibus target.

fncisco, 2017-03-25:
On omnibus it was on PA8, but I think I can't move it because PD2 isn't a timer pin... The only way I get works the led strip, was wiring it to PA8 on micro's pad.

asaerong, 2017-03-25:
let me understand it, PA8 = pwm 5 and Led=PD2 and the intention is to map PA8 to Led replacing PD2 but cannot because it isn't a timer pin. The only way is to wire the led pin to PA8 directly to the chip pad is that correct?

fncisco, 2017-03-26:
[image]

cterraza, 2017-06-28:
Holy pachito!!

mikeller, 2017-06-28:
I think the configurable LED pins that were added recently should allow for this to be solved? @jflyper